toinsson/pyrealsense #1: Compatibility with librealsense 1.9.7

scaldara55, 2016-09-07:
I had a working version of pyrealsense/librealsense.  I noticed changes to both repositories and decided to pull and rebuild the latest librealsense and pull and rebuild pyrealsense.  When I go to run a prior python script that was working, it crashes the python kernel when doing the pyrs.start().

Have you tried it with the librealsense 1.9.7?

toinsson, 2016-09-13:
just tried with tag 1.9.7, installed from scratch and do not see any issue.. 
Is the camera working without this bindings? e.g. executing "librealsense/bin/cpp-tutorial-3-pointcloud" for example.

scaldara55, 2016-09-13:
I am trying to get this working under Mac OS X, ultimately later to Ubuntu 14.04.  I did a brand new clone of librealsense and pyrealsense.  I can run "librealsense/bin/cpp-tutorial-3-pointcloud" and it executes directly.  Below are the steps I've taken to get things compiled and installed.  When I do the start command in python, it errors out with Option unsupported.  Details below.
----------------
# Building on OS X El Capitan 10.11.6

git clone https://github.com/IntelRealSense/librealsense.git
git clone https://github.com/toinsson/pyrealsense.git
cd librealsense
git checkout v1.9.7

# Edit Makefile and comment out ldconfig in install and uninstall targets

make uninstall
make clean
make all
make install

cd ../pyrealsense

Edit setup.py
Change:
	if ostype != 'Linux' and ostype != 'Windows':
To:
	if ostype != 'Linux' and ostype != 'Windows'and ostype != 'Darwin':
	
Change:
	if ostype == 'Linux':
To:
	if ostype == 'Linux' or ostype == 'Darwin':

/usr/local/bin/python setup.py install

# Start up python, import pyrealsense, issue a start to R200

	7831c1c083a2:pyrealsense scaldara$ /usr/local/bin/python
	Python 2.7.12 (default, Jun 29 2016, 14:05:02) 
	[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
	Type "help", "copyright", "credits" or "license" for more information.
	>>> import pyrealsense as pyrs
	>>> pyrs.start()
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 9 
	
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	1.000000 1.000000 100.000000 179.000000 100.000000 2.000000 16.000000 -1.000000 2500.000000 450.000000 1.000000 1.000000 6.000000 7.000000 -1.000000 
	rs_error was raised when calling rs_set_device_options(device:0x7f8f94813820, options:0x10d1d74d0, count:14, values:0x10d1d7948):
	    Option unsupported
	>>>

scaldara55, 2016-09-13:
Sorry about the "loud" text.  I saved my steps using OneNote and copied and pasted it in.  Lines with leading "#" got interpreted as Markdown headers.

toinsson, 2016-09-13:
ok, thanks for the additional info. I have added access to some parametrisation in the `start` method (effectively calling the function `create_context`). And it seems to be SR300 only ...

What you can try in the meantime is to comment out the call to `apply_ivcam_preset` at line 99 in realsense.c and reinstall `pyrealsense`. That should solve the problem. 

Or you can also try another `ivcam_preset` input value in the start call, like `pyrs.start(ivcam_preset=0)`. The range is between 0 and 8, see from line 518 onwards. (Now it is 9 as a default and it's custom parameters.) The thing is that I do not check the camera model in the `apply_ivcam_preset`, so this might not circumvent the issue.

 Will look how it's done in `librealsense` for possible update.

toinsson, 2016-09-14:
did it solve your issue ?

scaldara55, 2016-09-14:
Commenting out apply_ivcam_preset in realsense.c works.

Setting ivcam_preset to values 0-10 always results in Option unsupported.  The camera still basically functions and you can execute some of the other methods.

I don't understand the logic in apply_ivcam_preset() which checks for certain preset entries to be != -1 and then calls rs_set_device_options() with different args.

Should that logic check be "== -1"  ?

pyrs.start(ivcam_preset=0): 
	
	-----Executing the start() method -----
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 0
	rs_api_version: 10907
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	1.000000 1.000000 180.000000 303.000000 180.000000 2.000000 16.000000 -1.000000 1000.000000 450.000000 1.000000 1.000000 5.000000 1.000000 -1.000000
	rs_error was raised when calling rs_set_device_options(device:0x7fb7722b9020, options:0x1114ad440, count:14, values:0x1114ad480):
	    Option unsupported
	

pyrs.start(ivcam_preset=1)

	-----Executing the start() method -----
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 1
	rs_api_version: 10907
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	1.000000 0.000000 303.000000 605.000000 303.000000 -1.000000 -1.000000 -1.000000 1250.000000 975.000000 1.000000 1.000000 7.000000 0.000000 -1.000000
	rs_error was raised when calling rs_set_device_options(device:0x7f83f63d9820, options:0x11117a440, count:14, values:0x11117a4f8):
	    Option unsupported

pyrs.start(ivcam_preset=2)

	-----Executing the start() method -----
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 2
	rs_api_version: 10907
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	0.000000 0.000000 -1.000000 -1.000000 -1.000000 -1.000000 -1.000000 -1.000000 -1.000000 -1.000000 16.000000 1.000000 6.000000 2.000000 22.000000
	rs_error was raised when calling rs_set_device_options(device:0x7f9c75052020, options:0x10b12d440, count:15, values:0x10b12d570):
	    Option unsupported
	rs_error was raised when calling rs_set_device_options(device:0x7f9c75052020, options:0x10b12d440, count:14, values:0x10b12d570):
	    Option unsupported

pyrs.start(ivcam_preset=3)

	-----Executing the start() method -----
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 3
	rs_api_version: 10907
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	1.000000 1.000000 100.000000 179.000000 100.000000 2.000000 16.000000 -1.000000 1000.000000 450.000000 1.000000 1.000000 6.000000 3.000000 -1.000000
	rs_error was raised when calling rs_set_device_options(device:0x7faab4b44420, options:0x10feed440, count:14, values:0x10feed5e8):
	    Option unsupported

pyrs.start(ivcam_preset=4)

	-----Executing the start() method -----
[... 77 lines cut ...]
	-----Executing the start() method -----
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 9
	rs_api_version: 10907
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	1.000000 1.000000 180.000000 605.000000 303.000000 2.000000 16.000000 -1.000000 1250.000000 650.000000 1.000000 1.000000 5.000000 1.000000 -1.000000
	rs_error was raised when calling rs_set_device_options(device:0x7fb5149a4020, options:0x10a6f5440, count:14, values:0x10a6f58b8):
	    Option unsupported

pyrs.start(ivcam_preset=10)

	-----Executing the start() method -----
	c_height 480, c_width 640 c_fps 60
	d_height 480, d_width 640 d_fps 60
	depth_control_preset 0 ivcam_preset 10
	rs_api_version: 10907
	Using device 0, an Intel RealSense R200
	    Serial number: 2401005649
	    Firmware version: 1.0.72.06
	2.000000 0.000000 40.000000 1600.000000 800.000000 -1.000000 -1.000000 -1.000000 -1.000000 -1.000000 1.000000 -1.000000 -1.000000 -1.000000 -1.000000
	rs_error was raised when calling rs_set_device_options(device:0x7fe0e4000020, options:0x109345440, count:11, values:0x109345930):
	    Option unsupported

scaldara55, 2016-09-14:
Oops, did not want to close out the issue.

scaldara55, 2016-09-14:
Output of cpp-enumerate-devices from librealsense examples directory.

7831c1c083a2:examples scaldara$ ../bin/cpp-enumerate-devices 
Device 0 - Intel RealSense R200:
 Serial number: 2401005649
 Firmware version: 1.0.72.06
 USB Port ID: 20-6
 Supported options:
    COLOR_BACKLIGHT_COMPENSATION : 0 .. 4, 1, 1
    COLOR_BRIGHTNESS : 0 .. 255, 1, 56
    COLOR_CONTRAST : 16 .. 64, 1, 32
    COLOR_EXPOSURE : 1 .. 666, 1, 1
    COLOR_GAIN : 0 .. 256, 1, 32
    COLOR_GAMMA : 100 .. 280, 1, 220
    COLOR_HUE : 0 .. 0, 32767, 1.54391e+09
    COLOR_SATURATION : 0 .. 255, 1, 128
    COLOR_SHARPNESS : 0 .. 7, 1, 0
    COLOR_WHITE_BALANCE : 2000 .. 8000, 1, 6500
    COLOR_ENABLE_AUTO_EXPOSURE : 0 .. 1, 1, 1
    COLOR_ENABLE_AUTO_WHITE_BALANCE : 0 .. 1, 1, 1
    R200_LR_AUTO_EXPOSURE_ENABLED : 0 .. 1, 1, 0
    R200_LR_GAIN : 100 .. 6399, 1, 400
    R200_LR_EXPOSURE : 1 .. 330, 1, 164
    R200_EMITTER_ENABLED : 0 .. 1, 1, 0
    R200_DEPTH_UNITS : 0 .. 2.14748e+09, 1, 1000
    R200_DEPTH_CLAMP_MIN : 0 .. 65535, 1, 0
    R200_DEPTH_CLAMP_MAX : 0 .. 65535, 1, 65535
    R200_DISPARITY_MULTIPLIER : 1 .. 1000, 1, 32
    R200_DISPARITY_SHIFT : 0 .. 1, 1, 0
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_DECREMENT : 0 .. 255, 1, 5
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_INCREMENT : 0 .. 255, 1, 5
    R200_DEPTH_CONTROL_MEDIAN_THRESHOLD : 0 .. 1023, 1, 192
    R200_DEPTH_CONTROL_SCORE_MINIMUM_THRESHOLD : 0 .. 1023, 1, 1
    R200_DEPTH_CONTROL_SCORE_MAXIMUM_THRESHOLD : 0 .. 1023, 1, 512
    R200_DEPTH_CONTROL_TEXTURE_COUNT_THRESHOLD : 0 .. 31, 1, 6
    R200_DEPTH_CONTROL_TEXTURE_DIFFERENCE_THRESHOLD : 0 .. 1023, 1, 24
    R200_DEPTH_CONTROL_SECOND_PEAK_THRESHOLD : 0 .. 1023, 1, 27
    R200_DEPTH_CONTROL_NEIGHBOR_THRESHOLD : 0 .. 1023, 1, 7
    R200_DEPTH_CONTROL_LR_THRESHOLD : 0 .. 2047, 1, 24
    FRAMES_QUEUE_SIZE : 1 .. 20, 1, 20
 Stream DEPTH - 36 modes:
  640	x 480	@ 90Hz	Z16	56.2 x 43.6 degrees
  640	x 480	@ 90Hz	DISPARITY16	56.2 x 43.6 degrees
  640	x 480	@ 60Hz	Z16	56.2 x 43.6 degrees
  640	x 480	@ 60Hz	DISPARITY16	56.2 x 43.6 degrees
  640	x 480	@ 30Hz	Z16	56.2 x 43.6 degrees
  640	x 480	@ 30Hz	DISPARITY16	56.2 x 43.6 degrees
  628	x 468	@ 90Hz	Z16	55.3 x 42.6 degrees
  628	x 468	@ 90Hz	DISPARITY16	55.3 x 42.6 degrees
  628	x 468	@ 60Hz	Z16	55.3 x 42.6 degrees
  628	x 468	@ 60Hz	DISPARITY16	55.3 x 42.6 degrees
  628	x 468	@ 30Hz	Z16	55.3 x 42.6 degrees
  628	x 468	@ 30Hz	DISPARITY16	55.3 x 42.6 degrees
  492	x 372	@ 90Hz	Z16	55.8 x 43.6 degrees
  492	x 372	@ 90Hz	DISPARITY16	55.8 x 43.6 degrees
  492	x 372	@ 60Hz	Z16	55.8 x 43.6 degrees
  492	x 372	@ 60Hz	DISPARITY16	55.8 x 43.6 degrees
  492	x 372	@ 30Hz	Z16	55.8 x 43.6 degrees
  492	x 372	@ 30Hz	DISPARITY16	55.8 x 43.6 degrees
  480	x 360	@ 90Hz	Z16	54.7 x 42.4 degrees
  480	x 360	@ 90Hz	DISPARITY16	54.7 x 42.4 degrees
  480	x 360	@ 60Hz	Z16	54.7 x 42.4 degrees
  480	x 360	@ 60Hz	DISPARITY16	54.7 x 42.4 degrees
  480	x 360	@ 30Hz	Z16	54.7 x 42.4 degrees
  480	x 360	@ 30Hz	DISPARITY16	54.7 x 42.4 degrees
  332	x 252	@ 90Hz	Z16	55.7 x 43.7 degrees
  332	x 252	@ 90Hz	DISPARITY16	55.7 x 43.7 degrees
  332	x 252	@ 60Hz	Z16	55.7 x 43.7 degrees
  332	x 252	@ 60Hz	DISPARITY16	55.7 x 43.7 degrees
  332	x 252	@ 30Hz	Z16	55.7 x 43.7 degrees
[... 88 lines cut ...]
  628	x 468	@ 30Hz	Y16	55.3 x 42.6 degrees
  492	x 372	@ 90Hz	Y8	55.8 x 43.6 degrees
  492	x 372	@ 90Hz	Y16	55.8 x 43.6 degrees
  492	x 372	@ 60Hz	Y8	55.8 x 43.6 degrees
  492	x 372	@ 60Hz	Y16	55.8 x 43.6 degrees
  492	x 372	@ 30Hz	Y8	55.8 x 43.6 degrees
  492	x 372	@ 30Hz	Y16	55.8 x 43.6 degrees
  480	x 360	@ 90Hz	Y8	54.7 x 42.4 degrees
  480	x 360	@ 90Hz	Y16	54.7 x 42.4 degrees
  480	x 360	@ 60Hz	Y8	54.7 x 42.4 degrees
  480	x 360	@ 60Hz	Y16	54.7 x 42.4 degrees
  480	x 360	@ 30Hz	Y8	54.7 x 42.4 degrees
  480	x 360	@ 30Hz	Y16	54.7 x 42.4 degrees
  332	x 252	@ 90Hz	Y8	55.7 x 43.7 degrees
  332	x 252	@ 90Hz	Y16	55.7 x 43.7 degrees
  332	x 252	@ 60Hz	Y8	55.7 x 43.7 degrees
  332	x 252	@ 60Hz	Y16	55.7 x 43.7 degrees
  332	x 252	@ 30Hz	Y8	55.7 x 43.7 degrees
  332	x 252	@ 30Hz	Y16	55.7 x 43.7 degrees
  320	x 240	@ 90Hz	Y8	54 x 41.8 degrees
  320	x 240	@ 90Hz	Y16	54 x 41.8 degrees
  320	x 240	@ 60Hz	Y8	54 x 41.8 degrees
  320	x 240	@ 60Hz	Y16	54 x 41.8 degrees
  320	x 240	@ 30Hz	Y8	54 x 41.8 degrees
  320	x 240	@ 30Hz	Y16	54 x 41.8 degrees

toinsson, 2016-09-14:
glad it fixed it.

You can have a look in [rsutil.h](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rsutil.h), there's `rs_apply_ivcam_preset` which I used as a template.

I'll condition the function call on the camera model. That should be enough to fix the issue.

closed 2016-09-23 by scaldara55

ihabZhaika, 2016-11-08:
Hi,
i have the same issue , and i read the comments , but i did not figure out how you fixed it

```
c_height 480, c_width 640 c_fps 60
d_height 480, d_width 640 d_fps 60
depth_control_preset 0 ivcam_preset 9 

Using device 0, an Intel RealSense R200
    Serial number: 1431002230
    Firmware version: 1.0.72.06
1.000000 1.000000 100.000000 179.000000 100.000000 2.000000 16.000000 -1.000000 2500.000000 450.000000 1.000000 1.000000 6.000000 7.000000 -1.000000 
rs_error was raised when calling rs_set_device_options(device:0x10cd8f0, options:0x7ffd63928610, count:14, values:0x7ffd63928a88):
    Option unsupported
```

scaldara55, 2016-11-09:
In the file realsense.c, go to the "create_context" method and comment out:

    //apply_ivcam_preset(dev, ivcam_preset);

konnorbeard, 2016-11-29:
I'm trying (unsuccessfully) to get this up and running on OSX - I performed the same installation as @scaldara55 above and am able to install without any issues, and I'm able to run the examples in /librealsense/ without any problems. When I try to import pyrealsense, however, I get the following import error: 
```
>>> import pyrealsense
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Python/2.7/site-packages/pyrealsense/__init__.py", line 1, in <module>
    from pyrealsense import core
ImportError: dlopen(/Library/Python/2.7/site-packages/pyrealsense/core.so, 2): Library not loaded: lib/librealsense.so
  Referenced from: /Library/Python/2.7/site-packages/pyrealsense/core.so
  Reason: unsafe use of relative rpath lib/librealsense.so in /Library/Python/2.7/site-packages/pyrealsense/core.so with restricted binary
```

Can anyone shed some light on this? Seems like it has popped up in some other projects, but nothing seems to work. I've tried symlinking to the .so from different directories to no avail. 

Thanks!

toinsson, 2016-11-29:
was wondering if you could update the setup.py to point to your copy of the library, e.g. this 

    library_dirs = ["/usr/local/lib"],

This is where `librealsense.so` is stored under Ubuntu.
Otherwise, I have not been testing this on OSX but it seems to be working for at least one person.

konnorbeard, 2016-11-29:
That's where mine is stored as well: 

```
$ ls /usr/local/lib | grep realsense
librealsense.so
```

I see the include headers are in the right place: 
```
$ ls /usr/local/include/librealsense
rs.h		rs.hpp		rscore.hpp	rsutil.h
```

The "DepthSenseSDK" doesn't seem to exist though: 
```
$ ls /opt/softkinetic/DepthSenseSDK
ls: /opt/softkinetic/DepthSenseSDK: No such file or directory
```

The thing is that the module is being imported, it's just not picking up the wrapper methods:
```
>>> import pyrealsense as pyrs
>>> dir(pyrs)
['Device', 'Format', 'Stream', '__builtins__', '__doc__', '__file__', '__name__', '__package__', '__path__', 'core', 'enum']

>>> import inspect
>>> inspect.getmembers(pyrs, inspect.isfunction)
[]
```

Here's what I'm seeing in \_\_init\_\_.py, does this look correct? 
```
$ cat /Library/Python/2.7/site-packages/pyrealsense/__init__.py
from pyrealsense import core
from pyrealsense import enum

Device = core.Device
Format = enum.Format
Stream = enum.Stream
```

Thanks for your help with this.

konnorbeard, 2016-11-29:
Fixed this issue. I totally wiped the install/cache of pyrealsense off disk, reinstalled, and linked the binary differently. Here's the linked line that fixed it: 

```
$ sudo install_name_tool -change lib/librealsense.so /usr/local/lib/librealsense.so /Library/Python/2.7/site-packages/pyrealsense.so
```

Not sure why the error was showing up with core.so before - maybe I had conflicting packages with another pyrealsense that I tried? Working great now, thanks for the wrapper it's going to be really useful.