toinsson/pyrealsense #6: testing for ctypes branch

toinsson, 2016-12-07:
Hello guys,
since it seems you have been using the lib at some points, I was wondering if you could have a look at the ctypes branch. 
It is an update to the library that should make it easier to customise the camera streams (which ones as well as parameters). It should also work on Python 2 and 3.

I have been testing on my setup and it works ok so far. Wanted to have some feedback before pushing to master. Especially since some of you are using OsX and Windows, which I have not been trying out so far.

toinsson, 2016-12-07:
maybe @harajune, @scaldara55, @helxsz ?

closed 2016-12-13 by toinsson

scaldara55, 2017-04-05:
I just moved back to my project where I'm using pyrealsense.  I had done a far amount with the older C/C++ wrapper style.  CType is new to me and I'm just coming up to speed.  I would like to contribute where possible.  I'm a bit of a novice with open source git development.  I can do some testing, documentation and/or functional enhancements.  I've never done a "pull request" but would like to learn how.  It seems to me that the current pyrealsense only supports basic operations.  There are a lot more capabilities in librealsense that are not exposed in pyrealsense.  Do you have plans and time to support more capabilities?  Would you please suggest areas of the project that I might work on which would not duplicate things that you are planning on doing?