toinsson/pyrealsense #9: Accessing Multiple Cameras

joehager, 2017-01-04:
Hi,

I'm encountering an issue where if I try to access multiple cameras at the same time, I can only get frames from one of the units.

My system has two cameras attached, both are SR300 units. I'm using librealsense v 1.9.7.

I modified one of the example files as follows:

```
import logging
logging.basicConfig(level=logging.INFO)
import numpy as np
import cv2
import pyrealsense as pyrs

pyrs.start()
dev0 = pyrs.Device(device_id = 0, streams = [pyrs.ColourStream(fps=30)])
dev1 = pyrs.Device(device_id = 1, streams = [pyrs.ColourStream(fps=30)])

cnt = 0

while cnt < 30:
        dev0.wait_for_frame()
        dev1.wait_for_frame()
        cnt += 1

c0 = dev0.colour
c0 = cv2.cvtColor(c0, cv2.COLOR_RGB2BGR)

img_str = cv2.imencode('.jpg', c0)[1].tobytes()
file0 = open("file0.jpg", "wb")
file0.write(img_str)
file0.close()

c1 = dev1.colour
c1 = cv2.cvtColor(c1, cv2.COLOR_RGB2BGR)

img_str1= cv2.imencode('.jpg', c1)[1].tobytes()
file1 = open("file1.jpg", "wb")
file1.write(img_str1)
file1.close()
```

After executing this code, I see the same image in both files.

Am I using the python wrapper incorrectly? If so, what is the proper way to access two cameras?

Thanks,
Joe

toinsson, 2017-01-04:
yes, you are correct. 
I think it is because I set the property for the streams on the Device class and not the instance - see lines [102, 104, 109, 112]. However that is the recommended way to do for dynamic properties.
I'll look into it.

toinsson, 2017-01-04:
just made an update, it should be working now

joehager, 2017-01-05:
Yes, it's working now. Thanks!

closed 2017-01-05 by joehager