toinsson/pyrealsense #10: Can't install the library

kazoo-kmt, 2017-01-05:
When I tried to install (`sudo python setup.py install`), I got the followings. Do I need to copy rs.h by hand from librealsense?
```
running install
Checking .pth file support in /Library/Python/2.7/site-packages/
/usr/bin/python -E -c pass
TEST PASSED: /Library/Python/2.7/site-packages/ appears to support .pth files
running bdist_egg
running egg_info
writing pyrealsense.egg-info/PKG-INFO
writing top-level names to pyrealsense.egg-info/top_level.txt
writing dependency_links to pyrealsense.egg-info/dependency_links.txt
reading manifest file 'pyrealsense.egg-info/SOURCES.txt'
writing manifest file 'pyrealsense.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.12-intel/egg
running install_lib
running build_py
copying pyrealsense/constants.py -> build/lib.macosx-10.12-intel-2.7/pyrealsense
running build_ext
building 'pyrealsense.rsutilwrapper' extension
cc -fno-strict-aliasing -fno-common -dynamic -arch x86_64 -arch i386 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch x86_64 -arch i386 -pipe -I/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/numpy/core/include -I/usr/local/include/librealsense -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c pyrealsense/rsutilwrapper.c -o build/temp.macosx-10.12-intel-2.7/pyrealsense/rsutilwrapper.o
pyrealsense/rsutilwrapper.c:1:10: fatal error: 'rs.h' file not found
#include "rs.h"
         ^
1 error generated.
error: command 'cc' failed with exit status 1
```
I'm using macOS 10.12.2

toinsson, 2017-01-05:
Hello, you need to install librealsense before installing pyrealsense.

In the `cc` command from your log, there is `-I/usr/local/include/librealsense` which shows one include path. Under the directory `/usr/local/include/librealsense` you should find `rs.h`. If not, make sure librealsense is correctly installed.

And you shouldn't need sudo to install pyrealsense too.

I am not developing on Mac Os, so my help will be limited..

kazoo-kmt, 2017-01-05:
Thank you. librealsense is not under the directory `/usr/local/include/librealsense` in macOS's installation. I may need to place it manually.

kazoo-kmt, 2017-01-05:
When I ran `python setup.py install` without sudo, I got the following error:
```
running install
Checking .pth file support in /Library/Python/2.7/site-packages/
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/test-easy-install-26726.pth'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /Library/Python/2.7/site-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.

For information on other options, you may wish to consult the
documentation at:

  https://pythonhosted.org/setuptools/easy_install.html

Please make the appropriate changes for your system and try again.
```

When I put sudo and ran (after I copy pasted librealsense to `usr/loca/include`), I got another error as follows. Does this mean I'm still failing to install libreansense?
```
running install
Checking .pth file support in /Library/Python/2.7/site-packages/
/usr/bin/python -E -c pass
TEST PASSED: /Library/Python/2.7/site-packages/ appears to support .pth files
running bdist_egg
running egg_info
writing pyrealsense.egg-info/PKG-INFO
writing top-level names to pyrealsense.egg-info/top_level.txt
writing dependency_links to pyrealsense.egg-info/dependency_links.txt
reading manifest file 'pyrealsense.egg-info/SOURCES.txt'
writing manifest file 'pyrealsense.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.12-intel/egg
running install_lib
running build_py
copying pyrealsense/constants.py -> build/lib.macosx-10.12-intel-2.7/pyrealsense
running build_ext
building 'pyrealsense.rsutilwrapper' extension
cc -fno-strict-aliasing -fno-common -dynamic -arch x86_64 -arch i386 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch x86_64 -arch i386 -pipe -I/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/numpy/core/include -I/usr/local/include/librealsense -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c pyrealsense/rsutilwrapper.c -o build/temp.macosx-10.12-intel-2.7/pyrealsense/rsutilwrapper.o
In file included from pyrealsense/rsutilwrapper.c:2:
/usr/local/include/librealsense/rsutil.h:11:13: warning: unused function
      'rs_project_point_to_pixel' [-Wunused-function]
static void rs_project_point_to_pixel(float pixel[2], const struct rs_intrinsics * intrin...
            ^
/usr/local/include/librealsense/rsutil.h:55:13: warning: unused function
      'rs_transform_point_to_point' [-Wunused-function]
static void rs_transform_point_to_point(float to_point[3], const struct rs_extrinsics * e...
            ^
2 warnings generated.
In file included from pyrealsense/rsutilwrapper.c:2:
/usr/local/include/librealsense/rsutil.h:11:13: warning: unused function
      'rs_project_point_to_pixel' [-Wunused-function]
static void rs_project_point_to_pixel(float pixel[2], const struct rs_intrinsics * intrin...
            ^
/usr/local/include/librealsense/rsutil.h:55:13: warning: unused function
      'rs_transform_point_to_point' [-Wunused-function]
static void rs_transform_point_to_point(float to_point[3], const struct rs_extrinsics * e...
            ^
2 warnings generated.
cc -bundle -undefined dynamic_lookup -arch x86_64 -arch i386 -Wl,-F. build/temp.macosx-10.12-intel-2.7/pyrealsense/rsutilwrapper.o -L/usr/local/lib -lrealsense -o build/lib.macosx-10.12-intel-2.7/pyrealsense/rsutilwrapper.so
ld: library not found for -lrealsense
clang: error: linker command failed with exit code 1 (use -v to see invocation)
error: command 'cc' failed with exit status 1
```

kazoo-kmt, 2017-01-06:
Is the error above because of macOS? Or is it still able to fix?

kazoo-kmt, 2017-01-07:
This error was because of the incorrect installation. After trying the same process as Linux installation here (editorhttps://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#3rd-party-dependencies) at 5( provide a cmake file if you'd prefer to use your own favourite text ): 
```
mkdir build
cd build
cmake ..
make && sudo make install
```
the error vanished. However, I'm still having a same issue as https://github.com/toinsson/pyrealsense/issues/11.

kazoo-kmt, 2017-01-08:
This error might be because of the DLL's conflicts regarding Anaconda and was vanished after uninstalling Anaconda. Please close this if you confirm.

toinsson, 2017-01-08:
I guess on Mac the suffix to the lib file is not going to be so but dylib maybe ? 

you should change this line maybe:

    if _file.endswith(".so"):

kazoo-kmt, 2017-01-08:
I changed all '.so' to '.dylib' in __init__.py and utils.py. Then, I got the following error.
```
>>> import pyrealsense
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyrealsense/__init__.py", line 14, in <module>
    from pyrealsense.utils import pp, _check_error
  File "pyrealsense/utils.py", line 9, in <module>
    lrs = ctypes.CDLL('librealsense.dylib')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(librealsense.dylib, 6): image not found
```

toinsson, 2017-01-10:
I updated the library for Mac OS, you should be able to install it now. Please update to the latest commit on master (v1.2) and test again.

kazoo-kmt, 2017-01-11:
Thank you. I `git clone`d and `git checkout v1.2`, then ran `sudo python setup.py install`. I got followings when I tried to use `import pyrealsense`:
```
$ python
Python 2.7.10 (default, Jul 30 2016, 19:40:32) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.34)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyrealsense/__init__.py", line 14, in <module>
    from .utils import pp, _check_error
  File "pyrealsense/utils.py", line 7, in <module>
    from .importlib import lrs
  File "pyrealsense/importlib.py", line 25, in <module>
    rsutilwrapper = ctypes.CDLL(_find_extension_name())
  File "pyrealsense/importlib.py", line 23, in _find_extension_name
    return os.path.join(dirname, f_name)
UnboundLocalError: local variable 'f_name' referenced before assignment
```

toinsson, 2017-01-11:
You should import the library outside of the source directory, otherwise you won't see the compiled file.

kazoo-kmt, 2017-01-11:
I see. I tried from the home directory and got the followings.

```
$ python 
Python 2.7.10 (default, Jul 30 2016, 19:40:32) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.34)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Python/2.7/site-packages/pyrealsense-1.2-py2.7-macosx-10.12-intel.egg/pyrealsense/__init__.py", line 14, in <module>
    from .utils import pp, _check_error
  File "/Library/Python/2.7/site-packages/pyrealsense-1.2-py2.7-macosx-10.12-intel.egg/pyrealsense/utils.py", line 7, in <module>
    from .importlib import lrs
  File "/Library/Python/2.7/site-packages/pyrealsense-1.2-py2.7-macosx-10.12-intel.egg/pyrealsense/importlib.py", line 25, in <module>
    rsutilwrapper = ctypes.CDLL(_find_extension_name())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ctypes/__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/Library/Python/2.7/site-packages/pyrealsense-1.2-py2.7-macosx-10.12-intel.egg/pyrealsense/rsutilwrapper.so, 6): Library not loaded: librealsense.1.dylib
  Referenced from: /Library/Python/2.7/site-packages/pyrealsense-1.2-py2.7-macosx-10.12-intel.egg/pyrealsense/rsutilwrapper.so
  Reason: unsafe use of relative rpath librealsense.1.dylib in /Library/Python/2.7/site-packages/pyrealsense-1.2-py2.7-macosx-10.12-intel.egg/pyrealsense/rsutilwrapper.so with restricted binary
```

toinsson, 2017-01-11:
I'll advise you to use Anaconda's Python to start with. https://www.continuum.io/downloads
Create an environement, something like that:
```
conda create -n pyrs python=2
source activate pyrs
conda install numpy
pip install pycparser
```
And install the library with:
    
    python setup.py install

Otherwise, I am not sure I'd be able to provide you with more help : (

kazoo-kmt, 2017-01-12:
Thank you very much! I tried Anaconda and I could successfully install, though I needed to add `sudo` before `python setup.py install`(I got `error: [Errno 13] Permission denied: 'pyrealsense.egg-info/PKG-INFO'`)

kazoo-kmt, 2017-01-12:
Also, I would leave the note to install OpenCV (cv2) on this Anaconda environment to run the sample file. It's based on here (http://www.pyimagesearch.com/2016/11/28/macos-install-opencv-3-and-python-2-7/), but need some modification.
- `git clone https://github.com/opencv/opencv`
- `git clone https://github.com/opencv/opencv_contrib`
- `cd ~/opencv`
- `mkdir build`
- `cd build/`
-  `cmake -D CMAKE_BUILD_TYPE=RELEASE \`
     `-D CMAKE_INSTALL_PREFIX=/usr/local \`
     `-D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules \`
     `-D PYTHON2_LIBRARY=/Users/user_name/anaconda2/envs/pyrs/lib/libpython2.7.dylib \`
     `-D PYTHON2_INCLUDE_DIR=/Users/user_name/anaconda2/envs/pyrs/include/python2.7 \`
     `-D PYTHON2_EXECUTABLE=/Users/user_name/anaconda2/envs/pyrs/bin/python \`
     `-D BUILD_opencv_python2=ON \`
     `-D BUILD_opencv_python3=OFF \`
     `-D INSTALL_PYTHON_EXAMPLES=ON \`
     `-D INSTALL_C_EXAMPLES=OFF \`
     `-D BUILD_EXAMPLES=ON ..`
- `make -j4`
- `sudo make install`
- `cd /usr/local/lib/python2.7/site-packages/` (then check there is cv2.so in this directory)
- `cd /Users/mocchango/anaconda2/envs/pyrs/lib/python2.7/site-packages`
- `ln -s /usr/local/lib/python2.7/site-packages/cv2.so cv2.so`

closed 2017-01-12 by kazoo-kmt

ljc19800331, 2018-01-20:
Hello @kazoo-kmt, I am using ubuntu 16.04 to install realsense and get the similar problem. Can you show me the way of installing pyrealsense through anaconda? I tried many times but still get the error "pyrealsense/rsutilwrapper.c:1:10: fatal error: 'rs.h' file not found". Thanks.

kazoo-kmt, 2018-01-27:
Hi @ljc19800331, I followed the instruction here (https://github.com/toinsson/pyrealsense/issues/10#issuecomment-271952828) when installing with Anaconda. Can you find `rs.h` under `/usr/local/include/librealsense` as @toinsson  mentioned here (https://github.com/toinsson/pyrealsense/issues/10#issuecomment-270611635) before? If not, you might fail `librealsense`'s installation.

One thing you should notice is that `pyrealsense` doesn't support the latest `librealsense` as discussed here (https://github.com/toinsson/pyrealsense/issues/55). You need the legacy version of `librealsense` (https://github.com/IntelRealSense/librealsense/tree/v1.12.1).