toinsson/pyrealsense #11: OSError: librealsense.so: cannot open shared object file: No such file or directory

kazoo-kmt, 2017-01-07:
When I installed pyrealsense on Ubuntu 16.04 LTS and tried to run it, I had the following error. Have you ever had this error before?

```
>>> import pyrealsense as pyrs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyrealsense/__init__.py", line 14, in <module>
    from pyrealsense.utils import pp, _check_error
  File "pyrealsense/utils.py", line 8, in <module>
    lrs = ctypes.CDLL('librealsense.so')
  File "/home/username/anaconda2/lib/python2.7/ctypes/__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: librealsense.so: cannot open shared object file: No such file or directory
```

Just for your information, the followings are the log I got during the installation. It looks fine though I had some warnings. If the above error is because of the installation mistake, please let me know.

```
running install
running bdist_egg
running egg_info
creating pyrealsense.egg-info
writing pyrealsense.egg-info/PKG-INFO
writing top-level names to pyrealsense.egg-info/top_level.txt
writing dependency_links to pyrealsense.egg-info/dependency_links.txt
writing manifest file 'pyrealsense.egg-info/SOURCES.txt'
reading manifest file 'pyrealsense.egg-info/SOURCES.txt'
writing manifest file 'pyrealsense.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/pyrealsense
copying pyrealsense/__init__.py -> build/lib.linux-x86_64-2.7/pyrealsense
copying pyrealsense/stream.py -> build/lib.linux-x86_64-2.7/pyrealsense
copying pyrealsense/offline.py -> build/lib.linux-x86_64-2.7/pyrealsense
copying pyrealsense/utils.py -> build/lib.linux-x86_64-2.7/pyrealsense
copying pyrealsense/to_wrap.py -> build/lib.linux-x86_64-2.7/pyrealsense
copying pyrealsense/constants.py -> build/lib.linux-x86_64-2.7/pyrealsense
running build_ext
building 'pyrealsense.rsutilwrapper' extension
creating build/temp.linux-x86_64-2.7
creating build/temp.linux-x86_64-2.7/pyrealsense
gcc -pthread -fno-strict-aliasing -g -O2 -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/username/anaconda2/lib/python2.7/site-packages/numpy/core/include -I/usr/local/include/librealsense -I/home/username/anaconda2/include/python2.7 -c pyrealsense/rsutilwrapper.c -o build/temp.linux-x86_64-2.7/pyrealsense/rsutilwrapper.o
In file included from pyrealsense/rsutilwrapper.c:2:0:
/usr/local/include/librealsense/rsutil.h:11:13: warning: ‘rs_project_point_to_pixel’ defined but not used [-Wunused-function]
 static void rs_project_point_to_pixel(float pixel[2], const struct rs_intrinsic
             ^
/usr/local/include/librealsense/rsutil.h:55:13: warning: ‘rs_transform_point_to_point’ defined but not used [-Wunused-function]
 static void rs_transform_point_to_point(float to_point[3], const struct rs_extr
             ^
gcc -pthread -shared -L/home/username/anaconda2/lib -Wl,-rpath=/home/username/anaconda2/lib,--no-as-needed build/temp.linux-x86_64-2.7/pyrealsense/rsutilwrapper.o -L/usr/local/lib -L/home/username/anaconda2/lib -lrealsense -lpython2.7 -o build/lib.linux-x86_64-2.7/pyrealsense/rsutilwrapper.so
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/__init__.py -> build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/rsutilwrapper.so -> build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/stream.py -> build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/offline.py -> build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/utils.py -> build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/to_wrap.py -> build/bdist.linux-x86_64/egg/pyrealsense
copying build/lib.linux-x86_64-2.7/pyrealsense/constants.py -> build/bdist.linux-x86_64/egg/pyrealsense
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/stream.py to stream.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/offline.py to offline.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/utils.py to utils.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/to_wrap.py to to_wrap.pyc
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/constants.py to constants.pyc
creating stub loader for pyrealsense/rsutilwrapper.so
byte-compiling build/bdist.linux-x86_64/egg/pyrealsense/rsutilwrapper.py to rsutilwrapper.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrealsense.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrealsense.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrealsense.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying pyrealsense.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
pyrealsense.offline: module references __file__
pyrealsense.__init__: module references __file__
creating dist
creating 'dist/pyrealsense-1.2-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing pyrealsense-1.2-py2.7-linux-x86_64.egg
creating /home/username/anaconda2/lib/python2.7/site-packages/pyrealsense-1.2-py2.7-linux-x86_64.egg
Extracting pyrealsense-1.2-py2.7-linux-x86_64.egg to /home/username/anaconda2/lib/python2.7/site-packages
Adding pyrealsense 1.2 to easy-install.pth file

Installed /home/username/anaconda2/lib/python2.7/site-packages/pyrealsense-1.2-py2.7-linux-x86_64.egg
Processing dependencies for pyrealsense==1.2
Finished processing dependencies for pyrealsense==1.2
```

kazoo-kmt, 2017-01-08:
This error might be because of the DLL's conflicts regarding Anaconda and was vanished after uninstalling Anaconda. Please close this if you confirm.

toinsson, 2017-01-08:
can you put the output of: "file /usr/local/lib/librealsense.so".
I am wondering if you have a mismatch in architecture between librealsense and pyrealsense, similar to #8.

kazoo-kmt, 2017-01-08:
I already removed Anaconda, but the output of `file /usr/local/lib/librealsense.so` is `/usr/local/lib/librealsense.so: symbolic link to librealsense.so.1`

toinsson, 2017-01-10:
please check latest

closed 2017-01-10 by toinsson

tylerscheuble, 2017-01-29:
We are having this issue as well. We are on a fresh install of Ubuntu 16.04, and we have not installed anaconda. Our error is as follows:

```>>> import pyrealsense
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyrealsense-1.2-py2.7-linux-x86_64.egg/pyrealsense/__init__.py", line 14, in <module>
    from .utils import pp, _check_error
  File "/usr/local/lib/python2.7/dist-packages/pyrealsense-1.2-py2.7-linux-x86_64.egg/pyrealsense/utils.py", line 7, in <module>
    from .importlib import lrs
  File "/usr/local/lib/python2.7/dist-packages/pyrealsense-1.2-py2.7-linux-x86_64.egg/pyrealsense/importlib.py", line 25, in <module>
    rsutilwrapper = ctypes.CDLL(_find_extension_name())
  File "/usr/lib/python2.7/ctypes/__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: librealsense.so.1: cannot open shared object file: No such file or directory
```

Our output to `file /usr/local/lib/librealsense.so` is `/usr/local/lib/librealsense.so: symbolic link to librealsense.so.1.