toinsson/pyrealsense #13: Depth Image not working

kilsen512, 2017-01-26:
| Required Info  |   |
|---|---|
| Camera Model |SR300 | 
| Firmware Version | librealsense from source  | 
| Operating System & Version | Ubuntu 16.04  |

Hi,
I was trying your test/show_cv2.py, and I get the image for the colour image, however, the depth image is just red.
When I print the data, the array has the correct depth image size, but only two values are set (first two values)
array([[ 8288, 17669,     0, ...,     0,     0,     0],
       [    0,     0,     0, ...,     0,     0,     0],
       [    0,     0,     0, ...,     0,     0,     0],
       ..., 
       [    0,     0,     0, ...,     0,     0,     0],
       [    0,     0,     0, ...,     0,     0,     0],
       [    0,     0,     0, ...,     0,     0,     0]], dtype=uint16)

Thanks

PS: I had opened an issue regarding pyrealsense on windows, I was not able to solve it and switched to ubuntu. There seems to be a general problem regarding the compilation of plugins for python3 on windows.

toinsson, 2017-01-26:
hello, 
are you sure the camera is properly installed ? 
If not, check the examples from librealsense like cpp-tutorial-2-streams.

What version of python are you using?

ps: indeed, this package has no support for Windows.

kilsen512, 2017-01-27:
Hi, you were right. Somehow it wasn't properly installed.
Thank you!
[#312 from librealsense](https://github.com/IntelRealSense/librealsense/issues/312)

closed 2017-01-27 by kilsen512