toinsson/pyrealsense #14: How can I access librealsense C API functions?

nwestbury, 2017-02-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0 | 
| Operating System & Version | Ubuntu 16.04.1 LTS |

Silly question. How do I access `rs_get_device_extrinsics` from the [librealsense C API](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rs.h)? It seems like this was directly possible [at one point](https://github.com/toinsson/pyrealsense/commit/0d298642d382167515e5ca20194bc6e016982f52). Thanks for making this nifty wrapper!

toinsson, 2017-02-07:
Indeed. It's been removed .. I'll put that back in the next release, should be this week hopefully.

toinsson, 2017-02-13:
you should have a look at latest:

    def get_device_extrinsics(self, from_stream, to_stream):
        """Retrieve extrinsic transformation between the viewpoints of two different streams."""
        _rs_extrinsics = rs_extrinsics()
        lrs.rs_get_device_extrinsics(
            self.dev,
            from_stream,
            to_stream,
            ctypes.byref(_rs_extrinsics),
            ctypes.byref(e))
        _check_error(e)
        return _rs_extrinsics

This should do the trick. 

    e = dev.get_device_extrinsics(pyrs.rs_stream.RS_STREAM_COLOR, pyrs.rs_stream.RS_STREAM_DEPTH)

nwestbury, 2017-02-13:
Yeah that works. Thanks!

closed 2017-02-13 by nwestbury

toinsson, 2017-02-17:
by the way, 
not all librealsense functionality is wrapped in pyrealsense. But for most part it would be trivial to add a method on the DeviceBase class.
On top of that, you can also make direct call through `pyrs.lrs`.