toinsson/pyrealsense #15: question: Windows compatibility?

ZacDiggum, 2017-02-09:
Hi there, not an issue but a question:
Since I'm thinking about getting me a SR300 and develop a Python app for Windows, I wonder why your wrapper shouldn't work under Windows.
librealsense supports Windows 10 and Python is platform independent. So chances are good that it works, right? Your opinion on this would be very much appreciated. Or maybe you've heard of someone using the wrapper under Windows?

toinsson, 2017-02-09:
Hello, 
there's no major impediments in supporting windows. 

A couple of places that need to be changed:
- pyrealsense/setup.py:10 by passing the install directory as a parameter to the setup script.
- pyrealsense/pyrealsense/importlib.py:6/7 uncomment the windows suffix

And enabling the continuous testing with appveyor.
Plus some debugging, you never know what might break, but it should be minimal anyway.

I do not use a windows machine daily and so will not try to develop the support myself. However, if you feel up to the task, please go ahead and I'll provide some support and comments on pull requests.

ZacDiggum, 2017-02-09:
Thanks for the quick answer and also for the offer of help! I'll let you know how it's going when my realsense adventure has started. :-)
If that's okay for you I'd close this issue now but reopen it when there are things related to Windows.

ZacDiggum, 2017-02-11:
Hello @toinsson ,
here's another question that just occured to me:
In the SR300's specs it says a 6th gen. Intel CPU is required. Is that still true for librealsense + pyrealsense (under Ubuntu)? Are the RealSense products really vendor locked?
Thanks in advance for your help...

toinsson, 2017-02-13:
that's a question you should ask at RealSense directly : )

closed 2017-02-13 by toinsson

Jowizo, 2017-04-28:
Hey Toinsson, thanks for this wrapper!! 

I have a noobie question too, how can i change settings like adjusting the exposure in a python script?