toinsson/pyrealsense #19: SR300 crashed with: Timeout waiting for frames.

Chist777Porto, 2017-04-01:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| 3.15.0.0 |   | 
| OSX Capitan 10.11.4   |
| librealsense version  6d78ed9   |   |
| pyrealsense version 1.4v|   |

[image]

toinsson, 2017-04-05:
hello, is that happening consistently? 
I have seen this crash before, but a reset of the camera (unplugging it) usually solve the problem.
And I am not sure if it is due to pyrealsense or librealsense. Maybe this issue exists also on their issue tracker.

Chist777Porto, 2017-04-05:
Yes it is something constant, I have not been able to get that picture and it goes off.
I do not know if I have to install any drivers.
I appreciate if you can help me.

toinsson, 2017-04-05:
ok, then you should start from the examples in librealsense library. Are they working ? If yes, then the problem is in pyrealsense, otherwise it's just an install issue.

scaldara55, 2017-04-05:
I'm running macOS Sierra 10.12.4.  I just started using the new ctypes version of Pyrealsense.  I did the install from PyPI and it gave me a Pyrealsense version of 11201.  I haven't figured out how to trace this back to the git tag.  The show_cv2.py script works for me.  I added one line to print out the Pyrealsense API version.  When run I get a window with two video images, the left is RGB and the right is depth.  Click the mouse in the window and typing q exits.  I can run this multiple times without needed to unplug/replug the USB connector.

(pyrs2.7) 7831c1c083a2:pyrealsense-examples scaldara$ python my_show_cv2.py
INFO:pyrealsense:There are 1 connected RealSense devices.
INFO:pyrealsense:Using device 0, an Intel RealSense SR300
INFO:pyrealsense:    Serial number: 616205004463
INFO:pyrealsense:    Firmware version: 3.15.0.0
PyRealSense: 11201
(pyrs2.7) 7831c1c083a2:pyrealsense-examples scaldara$

toinsson, 2017-04-19:
any news about running the examples from librealsense @Chist777Porto ?

toinsson, 2017-05-12:
closed since it doesn't look like a standing issue.

closed 2017-05-12 by toinsson