toinsson/pyrealsense #21: Occasional segfault when calling pyrs.stop

zacstewart, 2017-04-13:
| Required Info  |   |
|---|---|
| Camera Model | SR300 (model VF0810) | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | macOS Sierra 10.12.3  |
| librealsense version | 1.12.1  |
| pyrealsense version | 1.4  |

Occasionally, my program crashes when stopping the device.
```
python(73262,0x7fffcfb4a3c0) malloc: *** error for object 0x7ffc237b89a8: incorrect checksum for freed object - object was probably modified after being freed.
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

Here's a stack trace from crash report:

```
Thread 11 Crashed:: org.libusb.device-hotplug
0   libsystem_c.dylib             	0x00007fffc6d914db __abort + 172
1   libsystem_c.dylib             	0x00007fffc6d9142f abort + 144
2   libsystem_malloc.dylib        	0x00007fffc6e8bf21 szone_error + 626
3   libsystem_malloc.dylib        	0x00007fffc6e80c44 tiny_malloc_from_free_list + 1148
4   libsystem_malloc.dylib        	0x00007fffc6e7f4ca szone_malloc_should_clear + 400
5   libsystem_malloc.dylib        	0x00007fffc6e7f2da malloc_zone_malloc + 107
6   com.apple.CoreFoundation      	0x00007fffb175f1d3 CFBasicHashCreateCopy + 131
7   com.apple.CoreFoundation      	0x00007fffb177404b CFSetCreateCopy + 123
8   com.apple.CoreFoundation      	0x00007fffb17ea17e __CFRunLoopRemoveAllSources + 142
9   com.apple.CoreFoundation      	0x00007fffb17741f2 __CFSetApplyFunction_block_invoke + 18
10  com.apple.CoreFoundation      	0x00007fffb17604ba CFBasicHashApply + 122
11  com.apple.CoreFoundation      	0x00007fffb1774199 CFSetApplyFunction + 185
12  com.apple.CoreFoundation      	0x00007fffb17ea23f __CFRunLoopRemoveAllSources + 335
13  com.apple.CoreFoundation      	0x00007fffb17e803f __CFFinalizeRunLoop + 303
14  com.apple.CoreFoundation      	0x00007fffb17e7c64 __CFTSDFinalize + 116
15  libsystem_pthread.dylib       	0x00007fffc6f1650f _pthread_tsd_cleanup + 544
16  libsystem_pthread.dylib       	0x00007fffc6f16249 _pthread_exit + 152
17  libsystem_pthread.dylib       	0x00007fffc6f16da8 pthread_exit + 30
18  libusb-1.0.0.dylib            	0x00000001109112d5 darwin_event_thread_main + 549
19  libsystem_pthread.dylib       	0x00007fffc6f14aab _pthread_body + 180
20  libsystem_pthread.dylib       	0x00007fffc6f149f7 _pthread_start + 286
21  libsystem_pthread.dylib       	0x00007fffc6f141fd thread_start + 13

Thread 11 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x000070000f219810  rcx: 0x00007fffc6e2d652  rdx: 0x0000000000000000
  rdi: 0x0000000000000003  rsi: 0x000070000f219810  rbp: 0x000070000f219820  rsp: 0x000070000f219800
   r8: 0x0000000000000000   r9: 0x0000000000989680  r10: 0x000070000f21a048  r11: 0x0000000000000246
  r12: 0x0000000000000001  r13: 0x000000010f0a9000  r14: 0x0000000000000000  r15: 0x000000010ee77000
  rip: 0x00007fffc6d914db  rfl: 0x0000000000010246  cr2: 0xffffff81c2f71004
  
Logical CPU:     2
Error Code:      0x00000000
Trap Number:     6
```

zacstewart, 2017-04-14:
A minimal reproduction of this bug:

```python
import pyrealsense as pyrs
import time

i = 0
while True:
    print("Iteration", i)
    pyrs.start()
    time.sleep(.25)
    pyrs.stop()
    time.sleep(.25)
    i += 1
```

```
$ python test.py                                                        │
Iteration 0                                                                                                            │
Iteration 1                                                                                                            │
Iteration 2                                                                                                            │
Iteration 3                                                                                                            │
Iteration 4                                                                                                            │
python(99336,0x7fffcfb4a3c0) malloc: *** error for object 0x7fc4f6a20298: incorrect checksum for freed object - object │
was probably modified after being freed.                                                                               │
*** set a breakpoint in malloc_error_break to debug                                                                    │
Abort trap: 6
```

toinsson, 2017-04-19:
yes, could reproduce the error. However, this does not seem to occur on Ubuntu. 
Maybe some platform dependent code is needed at the exit.

closed 2020-08-17 by toinsson