toinsson/pyrealsense #22: Memory leak: application memory grows continuously

massimilianoC, 2017-04-18:
Hi, i'm trying to optimize my code, but application memory grows at each cycle. Is there any function or way to optimize memory?
Thanks

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.71.06  | 
| Operating System & Version |  ubuntu 16.04 |
| librealsense version | 1.12.1  |
| pyrealsense version | 1.4  |

toinsson, 2017-04-19:
Could you be more specific about the application you are running ? 
The "leak" could be in your application and not in pyrealsense for example, or simply due to the way Python is doing garbage collection.

massimilianoC, 2017-04-19:
Hi, i have problems even with the example code: [show_cv2.py](https://github.com/toinsson/pyrealsense/blob/master/examples/show_cv2.py). 
Specifically i've misured about 190MB in 15 minutes of memory consumption running that sample code. And it's growing almost in a linear way through time, till the computer is overloaded.
I'm using that example structure for my application, so if i can optimize the example, i'm sure i'll have benefits even in my application.
Any suggestion?
Thanks for your time and for your work!!

toinsson, 2017-04-20:
here is a snapshot of the memory usage with `show_cv2.py` for 20+ minutes under python2. It seems stable with massive fluctuation which might be explained by the way the garbage collection works.

[image]

Python3 seems to be more performant in terms of memory footprint, might be good to test out.

massimilianoC, 2017-04-20:
Thanks. Python 3 seems to solve my problem! Thanks again

closed 2017-04-20 by massimilianoC