toinsson/pyrealsense #24: There seems to be a naming difference between this and librealsense

sonyccd, 2017-04-21:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | 2.0.71.28  | 
| Operating System & Version |  Mac OSX 10.12.4 |
| librealsense version | 1.12.1  |
| pyrealsense version | 1.4  |

For example dev.colour and dev.wait_for_frame(). In the librealsense doc it is color and wait_for_frames. With no list of the method names used here it is hard to work with this library. Am I missing something?

sonyccd, 2017-04-21:
Oh I see it is in the __init__.py file

toinsson, 2017-04-25:
Most of the API is located in the __init__.py file. You can also explore available functions by opening a python shell and tabing, "?"-ing on methods and variables. This is pop the docstring.

There is a branch in the making that provides a "readthedocs" website. It is not finished yet but will documented the API.

For colour vs color, I went with the UK english and wait_for_frame vs wait_for_frames, thought it made more sense. I agree that the API might benefit from sticking closely to the original librealsense namings.

toinsson, 2017-05-29:
I have changed the API to follow librealsense, and more code is in core.py now as well as some documentation on readthedocs. Hope that solves some issues.

closed 2017-05-29 by toinsson