toinsson/pyrealsense #25: Seems to be an issue with vc

PMARINA, 2017-04-24:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | DOESN'T MATTER YET
| Firmware Version |  NO IDEA | 
| Operating System & Version |  WINDOWS 10 |
| librealsense version |  NONE |
| pyrealsense version |  NONE |

Trying to install this using both the setup.py and through the pip way. 

Here is the output from the setup.py way (yes, I already downloaded dependencies):
```
C:\Users\XXXX\Downloads\pyrealsense-master>python setup.py install
running install
running bdist_egg
running egg_info
creating pyrealsense.egg-info
writing requirements to pyrealsense.egg-info\requires.txt
writing pyrealsense.egg-info\PKG-INFO
writing top-level names to pyrealsense.egg-info\top_level.txt
writing dependency_links to pyrealsense.egg-info\dependency_links.txt
writing manifest file 'pyrealsense.egg-info\SOURCES.txt'
reading manifest file 'pyrealsense.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pyrealsense.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-2.7
creating build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\constants.py -> build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\importlib.py -> build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\offline.py -> build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\stream.py -> build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\to_wrap.py -> build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\utils.py -> build\lib.win-amd64-2.7\pyrealsense
copying pyrealsense\__init__.py -> build\lib.win-amd64-2.7\pyrealsense
running build_ext
building 'pyrealsense.rsutilwrapper' extension
creating build\temp.win-amd64-2.7
creating build\temp.win-amd64-2.7\Release
creating build\temp.win-amd64-2.7\Release\pyrealsense
C:\Users\XXXX\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\amd64\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IC:\Python27\lib\site-packages\numpy\core\include -I/usr/local/include/librealsense -IC:\Python27\include -IC:\Python27\PC /Tcpyrealsense/rsutilwrapper.c /Fobuild\temp.win-amd64-2.7\Release\pyrealsense/rsutilwrapper.obj
rsutilwrapper.c
pyrealsense/rsutilwrapper.c(1) : fatal error C1083: Cannot open include file: 'rs.h': No such file or directory
error: command 'C:\\Users\\XXXX\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\amd64\\cl.exe' failed with exit status 2
```
Evidently, a file known as rs.h is missing, but I have no idea how to fix that. Maybe microsoft updated something recently that doesn't work with this installer?

toinsson, 2017-04-25:
Unfortunetly pyrealsense is not supporting Windows out of the box at the moment, however it is possible to adapt the code see #15 for short description.
If you are up to the task, I'll happily provide some guidance.

PMARINA, 2017-04-25:
I would really appreciate your help in installing this for windows. I looked through the other post, and I think I understand what I need to do.

naitikshukla, 2017-05-26:
Can you please tell me in 1st point on #15 what path i have to give and which location ?

toinsson, 2017-05-29:
there is a bit more documentation now

closed 2017-06-01 by toinsson