toinsson/pyrealsense #27: Windows Installation

randyyork123, 2017-05-13:
I've been trying to install this on Windows, which I know is not officially supported, but was hoping if I could receive some guidance or advice. I'm somewhat of a beginner but I'd love to help any way I can.

I notice there is currently a branch for Windows development that solved some of the initial issues I was having (namely, not being able to force C++ compilation due to .c extension, etc). I cloned that branch as is and am trying to work from there. I'm currently stuck here:

LINK : fatal error LNK1181: cannot open input file 'realsense.lib'
error: command '"C:\Users\XXXX\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\link.exe"' failed with exit status 1181

librealsense is installed and I verified it is working by going to the bin folder and running the .exe examples.
I've moved the complete librealsense folder (as downloaded and extracted from zip) to C:\RealSense and changed the setup.py file to point to "C:\RealSense\include\librealsense" which contains the .h files.

| Camera Model | R200 | 
| Operating System & Version | Windows 10 |
| pyrealsense version |  1.4 |

Thank you very much!

andrewramsay, 2017-05-13:
The version of setup.py in the windows branch relies on a couple of environment variables to determine the location of the librealsense files, since they could be almost anywhere on a Windows system. All you should need to do is set a variable called PYRS_INCLUDES to the folder containing the librealsense headers, and another called PYRS_LIBS to the folder containing realsense.lib. 

It sounds like you've bypassed the setting of PYRS_INCLUDES by editing setup.py directly, but you will still need to tell it where to find realsense.lib for the compilation to succeed.

randyyork123, 2017-05-13:
Thank you very much and please do pardon the newbie struggle!

Instead of modifying the setup file, I simply added the PYRS_INCLUDES and PYRS_LIBS environment variables, which allowed the setup to complete successfully.

When attempting to run the OpenCV example, I received the error "Unable to invoke 'cpp'.  Make sure its path was passed". This implies that pycparser cannot find cpp.exe. 

My first thought was to add this to my PATH, but I believe vcvarsall.bat can be run to set up the various environment variables automatically. This still did not help. I looked into the issue and it seems the Visual Studio C++ compiler command line executable is named cl.exe. I naively created a copy named cpp.exe and am now getting this error:

"cl : Command line warning D9024 : unrecognized source file type 'C:\RealSense\include\librealsense\rs.h', object file assumed"

I also tried compiling with MinGW, with an error related to this line: "rsutilwrapper = ctypes.CDLL(_find_extension_name())" and for some reason not being able to locate rsutilwrapper.pyd.

andrewramsay, 2017-05-14:
From a quick look at the [pycparser FAQ](https://github.com/eliben/pycparser#interaction-with-the-c-preprocessor), I'm not sure VC++ is the best choice as they recommend using other compilers on Windows. It might work but definitely not by simply renaming cl.exe. The parameters it expects will be totally different and you'd probably have to modify the way pyrealsense calls pycparser to fix that issue.

I would suggest that you go back to compiling pyrealsense with VC++ since that was working for you, and then just add the MinGW/bin directory to your PATH. It should contain a valid copy of cpp.exe and so allow things to work as they're supposed to do. This is the setup I have myself and it seems to work fine for me on a Win10 machine.

randyyork123, 2017-05-14:
Finally got it working, thanks for your help!

---
Some details:
I placed MinGW's bin folder in the PATH and successfully reinstalled the pyrealsense library. That being said, I continued to receive the error when attempting to import the pyrealsense library (or running the OpenCV example).

The error message states the module could not be located, pointing to this line:
rsutilwrapper = ctypes.CDLL(_find_extension_name()))

After some questioning (and cursing!), I ran Process Monitor (https://technet.microsoft.com/en-us/sysinternals/processmonitor.aspx) and realized that the error is caused by Python not being able to locate the previously-compiled realsense libraries. This is odd because the PATH is correctly set. 

For now, I placed the compiled libraries in the C:\Python27 folder and all is well.

closed 2017-05-14 by randyyork123

andrewramsay, 2017-05-15:
That does seem strange. If the .pyd file is in the same location as the rest of the pyrealsense source files and the folder containing realsense.dll is on the PATH then everything *should* just work. But good to hear it's working now anyway, thanks for testing it on Windows!

GiantDeveloper021, 2021-09-18:
Camera Model = R200 
Operating System = Windows 10
pyrealsense version =1.4

I hope you guys are doing well. when i install pyrealsense using pip install pyrealsense==1.4 having librealsense verison 1.12.1 it generates me following error;

pyrealsense/rsutilwrapper.c(1): fatal error C1083: Cannot open include file: 'rs.h': No such file or directory

but when i install using pip install pyrealsense then it generates following error;

LINK : fatal error LNK1181: cannot open input file 'realsense.lib'

i have set environment variables in users that are PYRS_INCLUDES AND PYRS_LIBS and given the path as shown in the pic below;

[image]

I will be very thankful if you guys could help me. Thanks is advance!

toinsson, 2021-09-18:
Hello,
I have no access to a windows machine right now so won't be able to check solutions.

It might be useful to make sure the terminal you execute the commands from has indeed both variables (PYRS_INCLUDES, PYRS_LIBS) set. You could check with `echo %VAR%`.

The problem you encounter is clearly a variable issue.

GiantDeveloper021, 2021-09-19:
Thanks for replying. i checked the variables using echo and they both outputted the same path as given in environment variable. See the screenshot below;

[image]

andrewramsay, 2021-09-19:
I think you have a couple of different problems here:

 * If I remember correctly PYRS_LIBS should be set to the directory containing realsense.lib, not the full path to the file itself
 * Is there a reason you're trying to use v1.4? That version seems to be too old to have had the Windows workaround with the 2 environment variables in setup.py, that was in the [2.0 release](https://github.com/toinsson/pyrealsense/releases/tag/v2.0). You can see the v1.4 version [here](https://github.com/toinsson/pyrealsense/blob/v1.4/setup.py) for reference. If you really need to use that specific version you'll have to add the workaround back in yourself and deal with any other problems that crop up
 * If the installer script is picking up your env vars, you should see that in the compiler output. For example your second screenshot doesn't show a `-IF:\FYP\Libraries\librealsense-1.12.1\include\librealsense` parameter when calling cl.exe, it should be there if the env var is visible to the setup.py script. Similarly when it tries to link to the .lib file at the next step you should be seeing a `/LIBPATH:F:\FYP\Libraries\librealsense-1.12.1\bin\Win32` parameter. In your screenshot they aren't there because of the v1.4 script not checking for the env vars, but if you try a newer release that should allow you to verify they're being detected correctly

GiantDeveloper021, 2021-09-19:
Thanks for your response:

- Ok now i have changed the path to **F:\FYP\Libraries\librealsense-1.12.1\bin\Win32** .
- No not really i was just giving it a try.
- After changing the path it's now showing both variables path but with new errors. This time i tried **pip install pyrealsense --user**.

[image]

andrewramsay, 2021-09-19:
Maybe an x64/x86 mismatch? It looks like you're using 64-bit Python here but the PYRS_LIBS path ends in "bin/Win32" which usually implies 32-bit binaries, so that could be why it's not resolving symbols correctly. If you have a "bin/x64" folder there with a realsense.lib inside I'd try that path instead.

GiantDeveloper021, 2021-09-19:
Thank you for the response! Ok i tried rebuilding the solution which created folder x64 in binary folder in which realsense.lib was created with some other .exe files and changed the path to x64 and then tried pip install pyrealsense --user but now it gives following error;

[image]

andrewramsay, 2021-09-19:
I don't think this is anything to do with pyrealsense at this point. It's successfully compiled and linked the module and is failing for some reason trying to run rc.exe which is the Visual Studio Resource Compiler tool. 

From a quick Google you may have hit [this bug](https://github.com/pypa/setuptools/issues/2212) which was fixed last year... but according to your screenshot you're using a version of pip that's about 3 years old, so you might have other outdated modules too! 

If this is a Python 3.6 installation you've been using for a long time you might want to start over with a fresh install of 3.8 or 3.9 to make sure things are properly up to date. At the very least I would try upgrading your versions of pip and setuptools.

GiantDeveloper021, 2021-09-25:
I uninstalled python 3.6 and installed the latest python 3.9 and updated pip too but it still shows me this ;

[image]
[image]

I missed the part that i am using visual studio 2015 update 5 as librealsense 1.12.1 was tested successfully on that. Before that , I tried to run librealsense 1.12.1 on visual studio 2019 but errors occured. 

Sorry due to some circumstances i couldn't reply back early.

andrewramsay, 2021-09-25:
As I said, I don't think this is a pyrealsense problem. The error is appearing during the compilation + linking of the C++ extension module, and that whole process is handled in a standard way by distutils/Cython. It's clearly picking up your env vars correctly which is the only pyrealsense-specific step in the whole process. 

I've just gone through the process of installing pyrealsense on a Win10 system: 
 * Python 3.9.5 x64
 * VS 2019 (v16.11.3)
 * Windows SDK v10.0.19041.0
 * MSVC build tools v142
 * librealsense-1.12.1

With that configuration it compiled + installed without any trouble. My guess is that there's something wrong with your Visual Studio setup. Have you tried installing any other Python modules that require compilation like this? 

If you [google that link.exe error code](https://www.google.com/search?q=link.exe+1158+python&hl=en) you can see a bunch of people reporting problems with Python modules and VS 2015 (aka 14.0) which is what you're using. If you look into the suggested solutions involving rc.exe and rcdll.dll I think that would be your best bet for fixing this.

GiantDeveloper021, 2021-09-25:
Ok so the good news is that pyrealsense-2.2 successfully installed with the help of this [solution](https://stackoverflow.com/questions/14372706/visual-studio-cant-build-due-to-rc-exe) but the bad news is that when i import pyrealsense it generates following error;

[image]

GiantDeveloper021, 2021-09-25:
> As I said, I don't think this is a pyrealsense problem. The error is appearing during the compilation + linking of the C++ extension module, and that whole process is handled in a standard way by [...]

yep i copy/paste both files into visual studio folder which solved the issue but now the problem is that i cannot import it as shown in the fig above . Thanks for the early reply

andrewramsay, 2021-09-25:
It's giving you an extremely clear error message that should have helped you search for solutions. There's even an [earlier post](https://github.com/toinsson/pyrealsense/issues/27#issuecomment-301256621) *in this same issue* where that exact problem is mentioned.

`Unable to invoke 'cpp'` ==> it tried to run a `cpp` executable and it can't find one. 

The exception is obviously coming from the pycparser module which has a [README that explains](https://github.com/eliben/pycparser#interaction-with-the-c-preprocessor) you will need a compatible binary available under Windows.

GiantDeveloper021, 2021-09-25:
I will look into it . Thank you for helping me!