toinsson/pyrealsense #30: _apply_ivcam_preset Function Not Found

randyyork123, 2017-05-24:
Hi there,

If I try to set an ivcam_preset like so: 
`c = pyrs.Device(ivcam_preset = 3)`

I'm getting an error:
> AttributeError: function '_apply_ivcam_preset' not found.

Librealsense is working otherwise, as are the associated examples. 
I can run pyrealsense OpenCV example and the streams are displayed.

Thanks very much!

| Camera Model | SR300 |
| Operating System & Version | Windows 10 |
| pyrealsense version | 1.4 |

toinsson, 2017-05-25:
Hello, 
I am guessing that you are using the dev branch ?
There is a problem with rsutilwrapper indeed, I'm looking into it.

randyyork123, 2017-05-25:
I was actually using the Windows branch. Thank you very much! 

For now, I have created a custom function that manually sets the particular options that are defined under the presets in librealsense. I know librealsense warns against this, but I think this is purely for maintaining uniformity and consistency (as it doesn't seem to be doing anything unique when the apply_ivcam_preset is called).

toinsson, 2017-05-29:
i've made some changes which should fix the issue. Feel free to try latest on master.

closed 2017-06-01 by toinsson