toinsson/pyrealsense #31: I am unable to perform import pyrealsense in Windows 10 64-bit with Python 2.7 in Anaconda

s0r2637, 2017-06-05:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| I don't know|   | 
| Windows 10, 64-bit |   |
| librealsense version is the latest from the source taken yesterday|   |
| pyrealsense version --  I can't find out because I can't import pyrealsense |   |

Hi,

I am unable to import pyrealsense.  I get the following error from the pycparser module. 
`RuntimeError: Unable to invoke 'cpp'.  Make sure its path was passed correctly
Original error: [Error 2] The system cannot find the file specified`

I don't get this problem in Ubuntu 16.04 LTS.  Please help, I need to make it work in Windows system

toinsson, 2017-06-06:
You'll need to provide more precise information. 

Can you describe what action produce the error message?
And how did you install pyrealsense?

s0r2637, 2017-06-08:
Hi,

It all works now.  The installation on Windows was a little tricky.  I had to correctly provide paths and that the rs.h and stdint.h have to be placed in the include folders under Anaconda's python distribution.  These small things were preventing me from successfully installing in Windows.  Thanks dude for a wonderful Python bindings to librealsense.

toinsson, 2017-06-08:
glad it works now. 

By the way, you should be able to provide the path information for rs.h via setting PYRS_INCLUDES environment variable.

closed 2017-06-08 by toinsson

ljc19800331, 2018-02-25:
Hello @s0r2637 , I face the similar problem.

I try to install pyrealsense2 both on python 2.7.11 and anaconda2 (with python 2.7.11). But I face the similar problem that "fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory". However, I don't know which 'stdint.h' file should I include in the system path. Where do you copy your 'stdint.h' file in your anaconda's python distribution? Would it possible if you provide more guidance for me? Thanks.

mpechoto, 2018-05-09:
I took stdin.h from C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.14.26428\include and paste to C:\Users\yourUser\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\include

mpechoto, 2018-05-09:
Hi @s0r2637 
What does the paths that you set?
@toinsson 
Please, let me know if it is correct:
PYRS_INCLUDES: C:\Users\mmarin\Documents\GitHub\librealsense-1.12.1\include\librealsense\
PYRS_LIBS C:\Users\mmarin\CMakeBuilds\90a93ced-a187-8e3f-b31b-522a0c28dbd7\build\x86-Debug

Thanks all

mpechoto, 2018-05-09:
I have the same issue when I just try to import pyrealsense:

import pyrealsense as pyrs

RuntimeErrorTraceback (most recent call last)
<ipython-input-1-efc124c40f81> in <module>()
----> 1 import pyrealsense as pyrs
      2 from pyrealsense.constants import rs_option
      3 
      4 import numpy as np
      5 from matplotlib import pyplot as plt

build\bdist.win32\egg\pyrealsense\__init__.py in <module>()

build\bdist.win32\egg\pyrealsense\core.py in <module>()

build\bdist.win32\egg\pyrealsense\constants.py in <module>()

C:\Users\mmarin\AppData\Local\Continuum\anaconda2\lib\site-packages\pycparser\__init__.pyc in parse_file(filename, use_cpp, cpp_path, cpp_args, parser)
     84     """
     85     if use_cpp:
---> 86         text = preprocess_file(filename, cpp_path, cpp_args)
     87     else:
     88         with open(filename, 'rU') as f:

C:\Users\mmarin\AppData\Local\Continuum\anaconda2\lib\site-packages\pycparser\__init__.pyc in preprocess_file(filename, cpp_path, cpp_args)
     47         raise RuntimeError("Unable to invoke 'cpp'.  " +
     48             'Make sure its path was passed correctly\n' +
---> 49             ('Original error: %s' % e))
     50 
     51     return text

RuntimeError: Unable to invoke 'cpp'.  Make sure its path was passed correctly
Original error: [Error 2] The system cannot find the file specified