toinsson/pyrealsense #37: build for windows

PrinceP, 2017-06-24:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version |  Windows 10 |
| librealsense version |   |
| pyrealsense version |   |

pyrealsense/rsutilwrapper.cpp(477): fatal error C1083: Cannot open include file: 'rs.h':

  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IP:\Prince\Anaconda3\envs\tensorflow-gpu\lib\site-packages\numpy\core\include -I/usr/local/include/librealsense -IP:\Prince\Anaconda3\envs\tensorflow-gpu\include -IP:\Prince\Anaconda3\envs\tensorflow-gpu\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\winrt" /EHsc /Tppyrealsense/rsutilwrapper.cpp /Fobuild\temp.win-amd64-3.5\Release\pyrealsense/rsutilwrapper.obj
  rsutilwrapper.cpp
  p:\prince\anaconda3\envs\tensorflow-gpu\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
  pyrealsense/rsutilwrapper.cpp(477): fatal error C1083: Cannot open include file: 'rs.h': No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

toinsson, 2017-06-25:
`Cannot open include file: 'rs.h': No such file or directory`

You need to set PYRS_INCLUDE environment variable to the directory where `rs.h` is located.

PrinceP, 2017-06-25:
Now I am getting below error:

pyrealsense/rsutilwrapperc.cpp(4): fatal error C1083: Cannot open include file: 'rsutil.h': No such file or directory

toinsson, 2017-06-25:
`rsutil.h` is at the same place as `rs.h`, under `xxx/librealsense/include/librealsense`.
What did you set `PYRS_INCLUDE` to ?

PrinceP, 2017-06-25:
PYRS_INCLUDE is correct now. But still this error. 

LINK : fatal error LNK1181: cannot open input file 'realsense.lib'

How to compile librealsense in windows ?

toinsson, 2017-06-25:
that's a question for `librealsense` i guess... : ) 
But I personally used VS2015 and compiled the library with the provided solution.

closed 2017-07-03 by toinsson