toinsson/pyrealsense #39: bad aligment between CAD stream and depth stream

arpit15, 2017-06-30:
| Required Info  |   |
| Camera Model | SR300 | 
| Firmware Version |  3.15.0.0 | 
| Operating System & Version | Ubuntu 14.04  |
| librealsense version |  1.12.1 |
| pyrealsense version |  master |

I was trying to visualize the CAD stream which I think means color aligned with depth and depth stream. I am getting pretty bad alignment between the 2 streams. Can somebody point me to the correct way to visualize aligned color to depth or depth to color?
[image]

toinsson, 2017-07-02:
hello, i am not sure I understand the issue, and if it is a `pyrealsense` related.. ?

arpit15, 2017-07-02:
I was just trying to see how the aligned depth and rgb image look using the pyrealsense library. They don't align. I don't know if its an issue in librealsense or pyrealsense?

toinsson, 2017-07-02:
keep in mind that the depth and color stream come from 2 different cameras, so they do not have the same point of view. You can see that clearly in the exmaple/cpp-tutorial-3-pointcloud. In white the depth point that don't have rbg information.

Still not sure that is the issue you are having here.

arpit15, 2017-07-02:
I am trying to overlay Depthstream and CADstream from pyrealsense package. I think CAD does the alignment of streams from rgb to depth stream and then projects that to depth frame. This is the visualization I have posted above. I don't see them aligned for sr300. 
The example file you are referring to in librealsense doesn't compile in my system due to some glw3 errors. Therefore I am not able to run that example from librealsense.
I wanted to help to get the aligned rgb and depth images in one coordinate frame either in depth camera frame or rgb camera frame.

toinsson, 2017-07-03:
So, cad gives you the color information of each depth pixel available in the depth stream. And dac gives you the depth information of each color pixel available in the color stream.

Can you explain what you mean by aligned, or what you are trying to achieve?

G-Paris, 2017-07-03:
I've had the same problem of DAC and color, or CAD end depth not aligning perfectly.

The cad and dac streams are the depth or color images multiplied with the intrinsic and extrinsic parameters of the camera sensors as determined by Intel.
The thing is that Intel does not calibrate each unique camera. All SR300's have the same parameter matrix. 
They store these parameters on the camera itself and librealsense reads this along with version# and camera typing.

So if you want better alignment, you would need to calibrate by hand.
You could achieve this with [OpenCV](http://docs.opencv.org/3.0-beta/doc/py_tutorials/py_calib3d/py_calibration/py_calibration.html).

In short; this is not a pyrealsense or librealsense issue, rather an Intel calibration problem.

toinsson, 2017-07-03:
awesome, thanks for the precision and the link!
I'll close the issue in that case.

closed 2017-07-03 by toinsson

arpit15, 2017-07-03:
I own 2 sr300 intel realsense camera which have different factory parameters. @G-Paris I have also tried opencv calibration. This doesn't result in good alignment as well. I was asking in pyrealsense forum because it was reported [here](https://github.com/intel-ros/realsense/issues/233#issuecomment-299490738) that if the camera parameters are not used properly then it might lead to bad alignment. Therefore I wanted to know if the camera parameters of the sensors are used properly with the pyrealsense lib.

G-Paris, 2017-07-03:
After noticing the DAC stream was misaligned, I researched how to calibrate it with opencv but in the end I did not re-calibrate the depth image because the DAC-stream was sufficient for my project at that time.

Also can't find the source where I read about the factory calibration, was about a half year ago. 

Pyrealsense calls the RS_STREAM_DEPTH_ALIGNED_TO_COLOR stream from the realsense library so the parameters should also be applied by the realsense library.

What you could still try is load the parameters from the camera and apply it yourself to the depth stream and see if you can make it align better?

arpit15, 2017-07-03:
@G-Paris I will try to use the parameters myself and check the result. Could you kindly post the results of the alignment you were getting with the factory settings?
Thanks