toinsson/pyrealsense #40: LR200 is not found

Wing0, 2017-07-01:
When I connect RealSense LR200 camera, no device is found. A RealSense error will be thrown with index out of range (see below). I am aware that LR200 is not supported with this library. However, as far as I understand, the difference to R200 is minimal. Is there a quick solution to get it to work?
```

File "realsense.py", line 11, in <module>
    pyrealsense.stream.DepthStream(fps=60)])
  File "/Users/vinkoo/.virtualenvs/cv30r/lib/python2.7/site-packages/pyrealsense/core.py", line 80, in Device
    _check_error(e)
  File "/Users/vinkoo/.virtualenvs/cv30r/lib/python2.7/site-packages/pyrealsense/utils.py", line 34, in _check_error
    pp(lrs.rs_get_error_message, e))
pyrealsense.utils.RealsenseError: rs_get_device(context:0x7fea0e222d40, index:0) crashed with: out of range value for argument "index"
```

| Required Info  |   |
|---|---|
| Camera Model | LR200 | 
| Firmware Version |  ? | 
| Operating System & Version | OS X El Capitan 10.11.6   |
| librealsense version | 1.12.1  |
| pyrealsense version | 2.0  |

toinsson, 2017-07-02:
hello, 
did you managed to use the LR200 with `librealsense`? 
If not, `pyrealsense` will not be able to connect any better, since it is just wrapping the code from `librealsense`.

see core.py:start:

```
    lrs.rs_create_context.restype = ctypes.POINTER(rs_context)
    ctx = lrs.rs_create_context(RS_API_VERSION, ctypes.byref(e))
    _check_error(e)

    n_devices = lrs.rs_get_device_count(ctx, ctypes.byref(e))
    logger.info("There are {} connected RealSense devices.".format(n_devices))
```

Wing0, 2017-07-02:
Good point, I will give it a try. Haven't tried yet.

toinsson, 2017-07-25:
any news ? Or can this be closed ?

Wing0, 2017-07-25:
Just came back from holidays and now back to the equipment. According to the documentation the librealsense should support LR200. However, I have not managed to test it yet. I try to organise time and report back in a week.

Thanks for following up.

toinsson, 2017-09-01:
please reopen if needed.

closed 2017-09-01 by toinsson