toinsson/pyrealsense #43: "failed to fetch an unpakcer" for non-native stream types

papr, 2017-07-06:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version | macOS Sierra 10.12.5  |
| librealsense version | 1.12.1 |
| pyrealsense version | 2.0 |
| python version | 3.6.0 |

I get an `RealsenseError: rs_start_device(): failed to fetch an unpakcer, most likely because enable_stream was not called!` when enabling only one non-native stream (e.g. PointStream). I need to enable at least one other native stream such that the non-native streams work.

Traceback:
```
Traceback (most recent call last):
  File "/Users/pabloprietz/work/test_scripts/realsense.py", line 6, in <module>
    with RS.Device(device_id=d['id'], streams=(s,)) as dev:
  File "/Users/pabloprietz/.virtualenvs/py3.6/lib/python3.6/site-packages/pyrealsense/core.py", line 136, in Device
    _check_error(e)
  File "/Users/pabloprietz/.virtualenvs/py3.6/lib/python3.6/site-packages/pyrealsense/utils.py", line 34, in _check_error
    pp(lrs.rs_get_error_message, e))
pyrealsense.utils.RealsenseError: rs_start_device(device:0x7fb80b18c620) crashed with: failed to fetch an unpakcer, most likely because enable_stream was not called!
```

Script to reproduce the error:
```python
import pyrealsense as RS
RS.start()
try:
    s = RS.stream.PointStream()
    for d in RS.core.get_devices():
        with RS.Device(device_id=d['id'], streams=(s,)) as dev:
            print(dev)
finally:
    RS.stop()
```

toinsson, 2017-07-07:
yes, the non-native streams are just wrapper around native streams, like the pointcloud is created from the depth map. Handling this automatically would require some logic (enabling required native streams based on input stream) which might introduce even more bugs : ) 

Since the error message is pretty clear, which makes it easy to understand and fix in application code - I am not sure if this should be implemented.

papr, 2017-07-07:
I understand. Is there documentation on which native streams each non-native stream is based on?

toinsson, 2017-07-07:
did not find it explicitly, but [rs.h](https://github.com/IntelRealSense/librealsense/blob/85652502139359895f48ac9704057bd9cfdd8923/include/librealsense/rs.hpp#L24) has some doc strings about each stream.

papr, 2017-07-07:
Ok, thank you.

closed 2017-07-07 by papr