toinsson/pyrealsense #44: ImportError: No module named 'Cython'

zacstewart, 2017-07-10:
Installing pyrealsense as a dependency to my own package is harder than it needs to be. Right now, I have to tell my users to do

```
pip install cython numpy
pip install .
```

Otherwise, when `setup.py` for pyrealsense is run, you get the following error:

```
Collecting pyrealsense==2.0 (from capture-tool==0.0.1)
  Using cached pyrealsense-2.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-_f4lmzo6/pyrealsense/setup.py", line 7, in <module>
        from Cython.Build import cythonize
    ImportError: No module named 'Cython'
```

It makes sense. Your `setup.py` imports from `Cython` first thing, but if the dependencies haven't been satisfied that won't work, and you can't satisfy the dependencies unless you can get past this import!

I did some digging on this issue, and it appears to be a well known problem that was solved with the release of [`distutils` 18.0](https://github.com/pypa/setuptools/blob/master/CHANGES.rst#180)

Another option is to [stub the `cythonize` function until it's available for import](https://stackoverflow.com/a/37471428/472486), but this is hacky and probably not necessary unless we have to support distutils < 18.0.

toinsson, 2017-07-10:
looks like v18.0 is pretty old though, so changing the code to let setuptools infer that it needs to cythonize the extension should be ok. If that works on all platform, it looks like an improvement to me. What do you think ?

zacstewart, 2017-07-11:
I agree. I think it's safe to rely on v18.0, and it would drastically simplify the install process.

However, we'd need to solve the same issue for Numpy to truly solve this problem. I see `np.get_include()` being called in setup.py, but that implies that numpy is already installed and available.

The gist is you have t let all the `setup_requires` be satisfied before importing any of them, but sometimes you need to import them in setup.py. You can dance around that in a few ways, like wrapping the import in `try` and returning a dummy function, or something like [this solution](https://stackoverflow.com/a/42163080/472486) that subclasses the `build_ext` class of distutils.