toinsson/pyrealsense #47: Uncaught c++ exception: VIDIOC_DQBUF error

papr, 2017-07-18:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version | Ubuntu 16.04 LTS |
| librealsense version | 1.12.1 |
| pyrealsense version | 2.0 |
| python version | 3.5.2 |

If the USB connection fails I get the following output and my Python thread dies:
```c++
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
```

Any idea how to catch this exception? Or should this be actually be caught by `librealsense`?

toinsson, 2017-07-19:
not sure I have seen that before. 
One thing though, `librealsense` + `pyrealsense` is not always stable when it comes to start/stop cycles. Sometimes plugging out the camera is needed to get the system back to normal.
So until one crash becomes systematic, I usually ignore them.

papr, 2017-07-19:
The crash is systematic and can be reproduced by disconnecting the camera while the device `is_streaming`.  This issue is related to https://github.com/IntelRealSense/librealsense/issues/418 and should probably not be handled by pyrealsense.

toinsson, 2017-07-19:
ok, didn't get that. Yes, no need to handle that in pyrealsense.

closed 2017-07-19 by papr