toinsson/pyrealsense #50: AttributeError: 'Service' object has no attribute 'Device'

xoryouyou, 2017-07-25:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.21.0.0 | 
| Operating System & Version | inux 4.11.9-1-ARCH   |
| librealsense version | librealsense.so.1.12.1     |
| pyrealsense version |  2.0  |

When running the "Online Usage" script from the `Readme.md` I get following error:

```
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    cam = serv.Device(device_id = 0, streams = [pyrs.stream.ColorStream(fps = 60)])
AttributeError: 'Service' object has no attribute 'Device'
```

After looking into the code of pyrealsense to see if there was any renaming or the sample is not up to date I tried in a REPL to see what the problem is:

```
Python 3.6.2 (default, Jul 20 2017, 03:52:27)
[GCC 7.1.1 20170630] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense as pyrs
>>> pyrs.Service
<class 'pyrealsense.core.Service'>
>>> serv = pyrs.Service()
>>> serv
<pyrealsense.core.Service object at 0x7f8e29121160>
>>> serv.get_devices
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Service' object has no attribute 'get_devices'
```

And it looks thats somehow the Service object is not a real Service object or is not initialized.

The librealsense SDK works and all their sample applications too, so I can rule that error-source out.

Any thoughts / ideas on this ?

papr, 2017-07-25:
You need pyrealsense 2.1 for this example to work.

xoryouyou, 2017-07-25:
@papr https://github.com/toinsson/pyrealsense/tree/v2.1  well is there any other location to get v2.1 ?

xoryouyou, 2017-07-25:
Google helped to find https://pypi.python.org/pypi/pyrealsense/2.1

Now it works, but why is 2.1 not on github when it's on pypi ?

papr, 2017-07-25:
The master branch should include the code for 2.1

toinsson, 2017-07-25:
I am pushing it as we speak : ) will update the tag on github

xoryouyou, 2017-07-25:
@toinsson thanks for https://github.com/toinsson/pyrealsense/releases/tag/v2.1
👍

closed 2017-07-25 by xoryouyou