toinsson/pyrealsense #51: Add YUVColorStream class

papr, 2017-07-25:
Tags: Enhancement, Feature Request

I would like to have a YUV color stream class but I do not know the appropriate parameters (`ndarray.dtype`, `ndarray.shape`, etc) to read out the data from librealsense.

@toinsson Would you be so kind and add an appropriate `YUVColorStream` to `streams.py`?

toinsson, 2017-07-25:
ah, thanks for the request : ) 

not sure how that would work though. Was looking around in `rs.h`, there is a `RS_FORMAT_YUYV`. However, I could not get it to work, unless I was passing the same info as with RGB.
e.g:

        self.stream = rs_stream.RS_STREAM_COLOR
        self.format = rs_format.RS_FORMAT_YUYV
        self.shape = (height, width, 3)
        self.dtype = ctypes.c_uint8

which returns a stream exactly equivalent to RGB.
There is some info about the formats:
https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf
and a comment about YUYV. Maybe you could ask or post something on `librealsense` github page?

Another way would be to convert RGB to YUV with something like: `cv2.cvtColor(c, cv2.COLOR_YUV2RGB)`

papr, 2017-07-26:
I am investigating this right now. I am creating a jupyter notebok that we can add to the examples later. I am actually able to get different data than RGB when using `RS_FORMAT_YUYV` but did not figure out the correct format yet.

Btw, would you mind writing me an email to the adress linked in my Github profile? I would like to open a communication channel that is not issue based. :)

closed 2017-09-26 by toinsson