toinsson/pyrealsense #57: Depth map of black pixels in an image is not coming

kishankedia, 2017-09-14:
Camera Model: SR300
Firmware Version: firmware': '3.21.0.0'
Operating System & Version: macOS sierra 10.12.6
librealsense version: 1.12.1
pyrealsense version: 2.1

**Required info:**

I am trying to get the depth map using intel realsense SR300. However i am not getting the depth map for the black pixels of an image. example: for my hair i am getting **0** value in the depth map.

[image]

toinsson, 2017-09-15:
Due to the way the SR300 camera work (by shining IR structured light onto the scene, recording it with an IR sensor and doing some processing on the recorded pattern to infer the depth), if a material in the scene does not reflect light - might be because it is reflective (glass, water, ...) or because it absorbs the light at that wavelength (red/black colour), it will show as either undefined or infinite depth.

This is what you see in your pictures. Nothing you can do here due to the nature of the sensor you are using.

closed 2017-09-15 by toinsson

kishankedia, 2017-09-15:
@toinsson can you suggest some depth camera that does not have such limitation and work in a sub-meter range. My friends tried using intel S200 camera and though the technology is almost same for both sr300 and s200, the results were drastically better for him than what i had

papr, 2017-09-15:
@kishankedia Before changing the devices, try to recalibrate your camera with the [Intel RealSense Camera Calibrator](https://downloadcenter.intel.com/download/24958/Intel-RealSense-Camera-Calibrator-for-Windows-).

kishankedia, 2017-09-15:
@papr  the link says that the camera calibrator is for R200 camera and not for SR300.