toinsson/pyrealsense #58: Using pyrealsense with Anaconda3 on Windows 10

fastreact92, 2017-09-19:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.3.27.5718 | 
| Operating System & Version | Windows 10 Education |
| librealsense version | 1.12.1 |
| pyrealsense version | 2.1 |

Hello, I'm senior grade student in South Korea and because of that I'm not good at using English well.

First, thank you very much for your support to using Intel RealSense on Python environment.
But, I have a problem in using pyrealsense.

I have librealsense 1.12.1 version and compiled examples successfully on VS2015 Update 3 version.
After compiled librealsense in two platform(Win32, x64), I got library files that named 'realsense-d.dll', 'realsense-d.exp', 'realsense-d.lib', 'realsense-d.lik', 'realsense-d.pdb' in both folders (librealsense's Directory)\bin\Win32, (librealsense's Directory)\bin\x64.

## (librealsense's Directory)\bin\Win32
[image]

## (librealsense's Directory)\bin\x64
[image]

And I added (librealsense's Directory)\include\librealsense in user environment named 'PYRS_INCLUDES' and I thought that the pyrealsense needs 'realsense.dll', 'realsense.lib', … . So I copied and changed name of that files and added bin's directory in user environment named 'PYRS_LIBS' too.

## PYRS_INCLUDES
[image]
[image]

## PYRS_LIBS
[image]

After that, I've installed pyrealsense successfully using Anaconda Prompt. But when I try to run example code(Online Usage), some errors occur like below.

## Running Online Usage code on Spyder 3
[image]

I've tried to fix this problem all day long but I couldn't fix it. Please help me anyone..

toinsson, 2017-10-20:
any progress? 
If you found the issue, please share it and close the issue.

fastreact92, 2017-10-20:
No, I haven't.. Have you any solutions..?

toinsson, 2017-10-24:
it looks like you cannot find the compiled library on your path.
There is a warning issued in the file `extlib.py` at line 26.

If you want to debug manually, you could run the code in a ipython shell and execute:
```
import sys
import warnings
import ctypes

os_name = sys.platform
lrs_prefix_mapping = {'darwin': 'lib', 'linux': 'lib', 'linux2': 'lib', 'win32': ''}
lrs_suffix_mapping = {'darwin': '.dylib', 'linux': '.so', 'linux2': '.so', 'win32': '.dll'}

try:
    lrs_prefix = lrs_prefix_mapping[os_name]
    lrs_suffix = lrs_suffix_mapping[os_name]
except KeyError:
    raise OSError('OS not supported.')

## import C lib
try:
    lrs = ctypes.CDLL(lrs_prefix+'realsense'+lrs_suffix)
import ctypes
import sys
import os
import warnings

os_name = sys.platform
lrs_prefix_mapping = {'darwin': 'lib', 'linux': 'lib', 'linux2': 'lib', 'win32': ''}
lrs_suffix_mapping = {'darwin': '.dylib', 'linux': '.so', 'linux2': '.so', 'win32': '.dll'}

try:
    lrs_prefix = lrs_prefix_mapping[os_name]
    lrs_suffix = lrs_suffix_mapping[os_name]
except KeyError:
    raise OSError('OS not supported.')

## import C lib
try:
    lrs = ctypes.CDLL(lrs_prefix+'realsense'+lrs_suffix)
except OSError:
    warnings.warn("librealsense not found.")
    lrs = None
```

On my machine, if I check the object `lrs` I get:
In [7]: lrs
Out[7]: <CDLL 'librealsense.dylib', handle 1048e5f80 at 0x106370320>

Another way to debug would be to check the path variable in your environment. 
In the ipython shell:
```
import os
print(os.environ['PATH'])
```
This should include the path that contains the library.

mnakazawa, 2017-10-24:
Hi.
I just started using pyrealsense, and faced the same issue on a similar environment (Win10, RealSense R200, librealsense1.21, pyrealsense2.2, etc.). 
I checked pyrealsense/extlib.py causing warnings, then I found pyrealsense was missing realsense.dll.

So I simply set environment variable PATH to a directory including realsense.dll.
I hope this will help you.

fastreact92, 2017-11-15:
Thank you all!! It works!

closed 2017-11-15 by toinsson