toinsson/pyrealsense #60: ImportError: cannot import name rs_stream

Chris-Lelievre, 2017-09-26:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |   | 
| Operating System & Version | Windows 10 Education 1607 |
| librealsense version | 10.0.26.0396  |
| pyrealsense version |  2.1 |

Believe computer is set up correctly but when I try to import pyrealsense I get the following error:

Traceback (most recent call last):
  File "\\st-james.ecs.vuw.ac.nz\lelievchri\Engr 489\python\unnumbered\pyrealsense_show_cv2.py", line 7, in <module>
    import pyrealsense as pyrs
  File "C:\Python27\lib\site-packages\pyrealsense\__init__.py", line 8, in <module>
    from .core import Service
  File "C:\Python27\lib\site-packages\pyrealsense\core.py", line 15, in <module>
    from .constants import RS_API_VERSION, rs_stream, rs_option
ImportError: cannot import name rs_stream

This is weird because pyrealsense works fine on a different identical computer. Any clues?

The script I ran is your show_cv2.py example but it also does this simply entering import pyrealsense in a command prompt.

Thanks heaps

papr, 2017-09-27:
@Chris-Lelievre Could you try running the same example using Python 3?

Chris-Lelievre, 2017-10-03:
When I try to install pyrealsense in python 3 I get the following error and I have no idea what it means, any help appreciated!

pip install pyrealsense
Collecting pyrealsense
 Using cached pyrealsense-2.2.tar.gz
Requirement already satisfied: numpy in c:\users\lelievchri\appdata\local\programs\python\python36-32\lib\site-packages (from pyrealsense)
Requirement already satisfied: cython in c:\users\lelievchri\appdata\local\programs\python\python36-32\lib\site-packages (from pyrealsense)
Requirement already satisfied: pycparser in c:\users\lelievchri\appdata\local\programs\python\python36-32\lib\site-packages (from pyrealsense)
Requirement already satisfied: six in c:\users\lelievchri\appdata\local\programs\python\python36-32\lib\site-packages (from pyrealsense)
Installing collected packages: pyrealsense
  Running setup.py install for pyrealsense ... error
    Complete output from command c:\users\lelievchri\appdata\local\programs\python\python36-32\python.exe -u -c "import setuptools, tokenize;\_\_file\_\_='C:\\Users\\LELIEV~1\\AppData\\Local\\Temp\\pip-build-9ha1rv6a\\pyrealsense\\setup.py';f=getattr(tokenize, 'open', open)(\_\_file\_\_);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, \_\_file\_\_, 'exec'))" install --record C:\Users\LELIEV\~1\AppData\Local\Temp\pip-znmszvxq-record\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-3.6
    creating build\lib.win32-3.6\pyrealsense
    copying pyrealsense\constants.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\core.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\extlib.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\extstruct.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\offline.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\stream.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\utils.py -> build\lib.win32-3.6\pyrealsense
    copying pyrealsense\__init__.py -> build\lib.win32-3.6\pyrealsense
    running build_ext
    building 'pyrealsense.rsutilwrapper' extension
    creating build\temp.win32-3.6
    creating build\temp.win32-3.6\Release
    creating build\temp.win32-3.6\Release\pyrealsense
    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\users\lelievchri\appdata\local\programs\python\python36-32\lib\site-packages\numpy\core\include -I/usr/local/include/librealsense "-IC:\Users\lelievchri\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\include" -Ic:\users\lelievchri\appdata\local\programs\python\python36-32\include -Ic:\users\lelievchri\appdata\local\programs\python\python36-32\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include\\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\\winrt" /EHsc /Tppyrealsense/rsutilwrapper.cpp /Fobuild\temp.win32-3.6\Release\pyrealsense/rsutilwrapper.obj
    rsutilwrapper.cpp
    c:\users\lelievchri\appdata\local\programs\common\microsoft\visual c++ for python\9.0\vc\include\codeanalysis\sourceannotations.h(68): warning C4467: usage of ATL attributes is deprecated
    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\vcruntime.h(92): warning C4005: '_CRT_STRINGIZE': macro redefinition
    C:\Users\lelievchri\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\include\crtdefs.h(180): note: see previous definition of '_CRT_STRINGIZE'
    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\vcruntime.h(95): warning C4005: '_CRT_WIDE': macro redefinition
    C:\Users\lelievchri\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\include\crtdefs.h(185): note: see previous definition of '_CRT_WIDE'
    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\vcruntime.h(156): warning C4005: '__CRTDECL': macro redefinition
    C:\Users\lelievchri\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\include\crtdefs.h(650): note: see previous definition of '__CRTDECL'
    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE\vcruntime.h(293): warning C4005: '_CRT_WARNING': macro redefinition
    C:\Users\lelievchri\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\include\crtwrn.h(24): note: see previous definition of '_CRT_WARNING'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\corecrt.h(457): error C2059: syntax error: 'constant'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\corecrt.h(457): error C2144: syntax error: 'int' should be preceded by ';'
    C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\corecrt.h(457): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
    C:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt\corecrt.h(457): warning C4183: '_Field_range_': missing return type; assumed to be a member function returning 'int'
[... cut ...]
Command "c:\users\lelievchri\appdata\local\programs\python\python36-32\python.exe -u -c "import setuptools, tokenize;\_\_file\_\_='C:\\Users\\LELIEV~1\\AppData\\Local\\Temp\\pip-build-9ha1rv6a\\pyrealsense\\setup.py';f=getattr(tokenize, 'open', open)(\_\_file\_\_);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, \_\_file\_\_, 'exec'))" install --record C:\Users\LELIEV\~1\AppData\Local\Temp\pip-znmszvxq-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\LELIEV\~1\AppData\Local\Temp\pip-build-9ha1rv6a\pyrealsense\

toinsson, 2017-10-20:
any progress? If you found the issue, please comment and close.

Chris-Lelievre, 2017-10-22:
Have not found the issue but the project I'm working on using this is coming to a close so I'm not actively looking to fix this myself. Feel free to close the issue if you want.
Cheers

closed 2017-10-23 by toinsson