toinsson/pyrealsense #61: Different

Mansterteddy, 2017-10-20:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| librealsense version |   |
| pyrealsense version |   |

toinsson, 2017-10-20:
Hello, 
could you provide some data? 
Otherwise it might be difficult to understand what you mean by "not as good" : ) 
It would be interesting to know whether the data is the same, or the difference comes from its visualisation.

Mansterteddy, 2017-10-21:
Hey, thanks for replying! I capture some results as follows.

For windows, I capture results with realsense SDK:
[image]

For mac, I capture results with your examples:
[image]
[image]

Can you tell me how to solve this issue?

toinsson, 2017-10-21:
the difference comes from the way you plot your data. 
This is not a pyrealsense issue. You will have to look into the visualisation code.

closed 2017-10-21 by toinsson