toinsson/pyrealsense #62: Can't install from source on Windows

dolhasz, 2017-10-24:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Windows 10 |
| librealsense version |  2.8 |
| pyrealsense version |  2.2 |

Hi there, I have been trying to install pyrealsense 2.2 from source and pip but keep getting following error. Followed the installation guide and added all required paths (I think, at least). 

Getting following error:

```
building 'pyrealsense.rsutilwrapper' extension
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ipyrealsense -IC:\ProgramData\Miniconda2\envs\tf_gpu\lib\site-packages\numpy\core\include -I/usr/local/include/librealsense "-IC:\Program Files (x86)\Intel RealSense SDK 2.0\include\librealsense2" -IC:\ProgramData\Miniconda2\envs\tf_gpu\include -IC:\ProgramData\Miniconda2\envs\tf_gpu\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.10240.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include\\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\\winrt" /EHsc /Tppyrealsense/rsutilwrapper.cpp /Fobuild\temp.win-amd64-3.5\Release\pyrealsense/rsutilwrapper.obj
rsutilwrapper.cpp
c:\programdata\miniconda2\envs\tf_gpu\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(3): error C2065: 'rs_device': undeclared identifier
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(3): error C2065: 'device': undeclared identifier
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(3): error C2062: type 'int' unexpected
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(4): error C2065: 'rs_device': undeclared identifier
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(4): error C2065: 'device': undeclared identifier
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(4): error C2065: 'rs_ivcam_preset': undeclared identifier
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(4): error C2146: syntax error: missing ')' before identifier 'preset'
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(6): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
c:\users\aland_000\google drive\python\cards\pyrealsense-2.2\pyrealsense\rsutilwrapper.h(6): error C2143: syntax error: missing ',' before '*'
pyrealsense/rsutilwrapper.cpp(1774): error C3861: '_apply_ivcam_preset': identifier not found
pyrealsense/rsutilwrapper.cpp(1956): error C3861: '_apply_ivcam_preset': identifier not found
pyrealsense/rsutilwrapper.cpp(2147): error C2660: '_project_point_to_pixel': function does not take 3 arguments
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2
```

What am I missing here?

Thanks

ntheru, 2017-10-26:
Hello, I believe that I have the same problem. From what I understand, there has been a change in the original librealsense folder. /usr/local/include/**librealsense2** <- as the original code looks for /usr/local/include/librealsense

ntheru, 2017-10-26:
if you can get the older version of librealsense, it will work. Tried and tested for 
pip install pyrealsense

toinsson, 2017-10-26:
this is correct, `pyrealsense` supports only `librealsense` v1.x. 
If you are using `librealsense` v2, there is a python wrapper already included in the library, 
see https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python.

closed 2017-10-26 by toinsson