toinsson/pyrealsense #64: Can't use on Linux kernel version after 4.4

kazoo-kmt, 2018-01-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04 LTS (kernel version 4.10)  |
| librealsense version | v1.12.1  |
| pyrealsense version | 2.2  |

When I tried to install pyrealsense with the legacy librealsense (v1.12.1, the latest version that we can use with pyrealsense) based on the manual (https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md), the installation fails when patching uvcvideo module (when running `./scripts/patch-uvcvideo-16.04.simple.sh`). This is because this patch isn't applicable for the kernel after 4.4 version. 

I think this is a known issue, but what is the recommendation when having this issue? Use the old kernel version or use pyrealsense2 from the latest librealsense?

toinsson, 2018-01-08:
This sounds like an issue for librealsense, not pyrealsense. Better head to their repo for troubleshooting.

svarnypetr, 2018-02-14:
I can confirm, this is a librealsense problem. Had to change my kernel also back to 4.4.

closed 2018-08-20 by toinsson