toinsson/pyrealsense #71: Is there a way to isntall pyrealsense2 in anaconda2 (python 2.7)

ljc19800331, 2018-02-24:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Windows 10  |
| librealsense version |  Latest |
| pyrealsense version | pyrealsense2  |

Hello,

I can successfully install pyrealsense2 in windows 10, but I can't find any information which can install with Anaconda2. Do I need to compile from scratch using conda in command line? Thanks.

toinsson, 2018-02-25:
`Cannot open include file: 'stdint.h'`

you need to have this file in your path, (mentioned in the prerequisites : ) ). Just make sure you download it or make it available.

closed 2018-02-25 by toinsson

ljc19800331, 2018-02-25:
Thanks for the reply.

I try to add the 'stdint.h' file ( which is in the location: C:\Anaconda2\Library\include and ) to the environment path. But this problem still exists.

"Make sure you download it or make it available" -- Do you mean download this single file or just download the microsoft visual c++ package for python?

I have successfully installed pyrealsense2 package in python 2.7.11 but not with anaconda ( I have installed both on my computer). Do I need to use the same 'stdint.h' file with the same in python 2.7.11?

I also want to try with the method: 
conda create -n pyrs python=2
source activate pyrs
conda install numpy
pip install pycparser

But I worry it will influence my original packages in anaconda2.

Thanks

ljc19800331, 2018-02-26:
I have solved the problem. Thanks for your guidance.