toinsson/pyrealsense #72: Error in Windows Installation: Unable to import pyrealsense

mapleee, 2018-03-16:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   | 
| Operating System & Version | Windows 10 |
| librealsense version | 1.12.1 |
| pyrealsense version | 2.2 |

[image]
Hello! 
     I have successfully installed the pyrealsense on Windows 10. However, when I try to import pyrealsense in python, I encounted the problem above. I have set the environ varible PYRS_INCLUDES to the path C:\pyrealsense\librealsense-1.12.1\include\librealsense, which has rs.h, stdint.h,etc. And I also set PYRS_LIBS to the path C:\pyrealsense\librealsense-1.12.1\bin\x64 which has realsense.dll and realsense.lib.  Please help, I need to make it work in Windows system.

toinsson, 2018-03-16:
It is a file not found error. Please double check `rs.h` can be accessed from python space.

toinsson, 2018-03-16:
In `constants.py` there is the code that look for the header file.

mapleee, 2018-03-17:
I tried your suggestion, however, I still face the same problem. 
Firstly, I'm sure that rs.h can be accessed from python space because if the program can't find the rs.h file, it will have another error(This is not my actuall error, I just move the rs.h file to a wrong place to test if the program can find it. Obviously, it's a path error):
[image]
Secondly, I print the variable rs_h_filename to resure the path:
[image]There is nothing wrong.
Consequently, I'm still in a mess and I still can't find the real problem.

toinsson, 2018-03-19:
seems like an issue in pycparser ? maybe you can try to execute the line:

`ast = pycparser.parse_file(rs_h_filename, use_cpp=True)`

by itself and debug from there.

(Also, make sure you are in another directory than pyrealsense when importing the library.)

closed 2018-03-29 by toinsson