toinsson/pyrealsense #74: switch between two IR camera

zxh009123, 2018-03-30:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | ubuntu 16.04 LTS  |
| librealsense version | 1.12.1  |
| pyrealsense version |  2.2 |

Hi,
I am currently using realsense r200 on a project and I need the infrared stream. As in the realsense r200 spec, I notice that there are two  IR cameras on the realsense r200. In pyrealsense, it use the right IR camera as default stream. Is there any way I can get the IR stream from the other IR camera?

toinsson, 2018-04-03:
if you look at:
https://github.com/toinsson/pyrealsense/blob/1b327915ee8e7a28d17c0446a720a59356fc8a40/pyrealsense/stream.py#L112
you'll notice: `self.stream = rs_stream.RS_STREAM_INFRARED`.

In the `rs_stream` structure there is also a `rs_stream.RS_STREAM_INFRARED2` which should point to the second IR stream.

So define new InfraredStream2 with the correct parameter and pass it to the Device init function.

zxh009123, 2018-04-04:
Hi toinsson,
    Thanks for your reply. This works very well!

closed 2018-04-04 by zxh009123