toinsson/pyrealsense #75: How to get the focal length? How to get the sensor height?

MathiasDePaepe, 2018-04-03:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.71.06 | 
| librealsense version | 1.x.x  |

Hello, i'm a student and i'm wondering how i can get the focal length of the R200 to determine the realtime dimensions of a detected object when i now the distance between the camera and the object.

I want to use the following equation:

realtime height [mm] = ( focal length [mm] * distance to object [mm] * image height [pixels] ) / ( object height [pixels] * sensor height [mm] )

Thanks on advance

toinsson, 2018-04-04:
you should have a look at the structure that contains the intrinsic parameters of the camera: https://github.com/IntelRealSense/librealsense/blob/8343c0c7dc6093ed65b2d687741f8f76977f2e34/include/librealsense/rs.h#L300
This structure is wrapped in the module `extstruct`. Check the documentation too:
http://pyrealsense.readthedocs.io/en/master/pyrealsense.html#pyrealsense.extstruct.rs_extrinsics

closed 2018-04-04 by toinsson

MathiasDePaepe, 2018-04-04:
When i'm compiling the following code:

rs::intrinsics intrin;
printf( "intrin: %s\n", intrin.hfov() );

I get the message: segmentation fault (core dumped)
When i'm changing the type of %s to %d i get a value of 0. When I use vfov() instead i get a value of 180.

Do you know the problem?