toinsson/pyrealsense #77: fatal error LNK1120: 2 unresolved externals

mpechoto, 2018-05-10:
| Required Info  |   |
|---|---|
| Camera Model | F200| 
| Firmware Version |   | 
| Operating System & Version | Windows 7  |
| librealsense version |  1.12.1 |
| pyrealsense version |  2.2 |

I am unable to perform import pyrealsense in Windows 7 64-bit with Python 2.7 in Anaconda
(Similar to issue #31)

I've opened the example pyrealsense/examples/show_jupyter.ipynb and tried to run. The import pyrealsense did not work:

RuntimeErrorTraceback (most recent call last)
<ipython-input-2-efc124c40f81> in <module>()
----> 1 import pyrealsense as pyrs
      2 from pyrealsense.constants import rs_option
      3 
      4 import numpy as np
      5 from matplotlib import pyplot as plt

build\bdist.win32\egg\pyrealsense\__init__.py in <module>()

build\bdist.win32\egg\pyrealsense\core.py in <module>()

build\bdist.win32\egg\pyrealsense\constants.py in <module>()

C:\Users\mmarin\AppData\Local\Continuum\anaconda2\lib\site-packages\pycparser\__init__.pyc in parse_file(filename, use_cpp, cpp_path, cpp_args, parser)
     84     """
     85     if use_cpp:
---> 86         text = preprocess_file(filename, cpp_path, cpp_args)
     87     else:
     88         with open(filename, 'rU') as f:

C:\Users\mmarin\AppData\Local\Continuum\anaconda2\lib\site-packages\pycparser\__init__.pyc in preprocess_file(filename, cpp_path, cpp_args)
     47         raise RuntimeError("Unable to invoke 'cpp'.  " +
     48             'Make sure its path was passed correctly\n' +
---> 49             ('Original error: %s' % e))
     50 
     51     return text

RuntimeError: Unable to invoke 'cpp'.  Make sure its path was passed correctly
Original error: [Error 2] The system cannot find the file specified

Thanks,

mpechoto, 2018-05-15:
Finally I could install and run the examples. I had to install MinGW in order to recognize cpp compiler.

closed 2018-05-15 by mpechoto

HanSangJun, 2019-02-01:
@mpechoto I cannot solve this problem, Could you explain more specifically? Why do we need another compiler?