toinsson/pyrealsense #81: Trying to install under Arch Linux x86_64

Kuchiriel, 2018-07-16:
| Required Info  |   |
|---|---|
| Camera Model | IDK - Not using right now | 
| Firmware Version |  IDK | 
| Operating System & Version |  Latest Arch Linux x86_64 |
| librealsense version |  IDK |
| pyrealsense version | 2.2  |

```
λ trevor [~] → sudo easy_install pyrealsense
Searching for pyrealsense
Reading https://pypi.org/simple/pyrealsense/
Downloading https://files.pythonhosted.org/packages/6b/08/bdf4ad92bffd0c53d403aa41af012d75315fb14896537c1e04e1c4396f72/pyrealsense-2.2.tar.gz#sha256=c9933eb16384283c3058e8b37d7c8677066412bfdb0ab31517030deafb094dce
Best match: pyrealsense 2.2
Processing pyrealsense-2.2.tar.gz
Writing /tmp/easy_install-ovwucddr/pyrealsense-2.2/setup.cfg
Running pyrealsense-2.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-ovwucddr/pyrealsense-2.2/egg-dist-tmp-zpk4ywt_
Compiling pyrealsense/rsutilwrapper.pyx because it depends on /usr/lib/python3.6/site-packages/Cython/Includes/numpy/__init__.pxd.
[1/1] Cythonizing pyrealsense/rsutilwrapper.pyx
In file included from /usr/lib/python3.6/site-packages/numpy/core/include/numpy/ndarraytypes.h:1816,
                 from /usr/lib/python3.6/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /usr/lib/python3.6/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from pyrealsense/rsutilwrapper.cpp:611:
/usr/lib/python3.6/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
pyrealsense/rsutilwrapper.cpp:614:10: fatal error: rs.h: No such file or directory
 #include "rs.h"
          ^~~~~~
compilation terminated.
error: Setup script exited with error: command 'gcc' failed with exit status 1
```

toinsson, 2018-07-16:
`pyrealsense/rsutilwrapper.cpp:614:10: fatal error: rs.h: No such file or directory
 #include "rs.h"`

This line means that the librealsense library is not found. Make sure you have it installed.

MahdiSMIDA, 2018-07-17:
same problem

toinsson, 2018-07-17:
I am not familiar with Arch Linux. Do you know where headers do go when you install a library?
Currently, the setup step expect them to be located under `/usr/local/include/librealsense`, see: https://github.com/toinsson/pyrealsense/blob/4a345a2fc9b5b5bb3ef9db3ac0aa27fd00aa3489/setup.py#L34.
@MahdiSMIDA, are you also having the same system?

Kuchiriel, 2018-07-18:
@toinsson In Arch it is located at `/usr/lib/modules`, Thanks for the enlightment!

Kuchiriel, 2018-07-18:
@toinsson Same location but I've seen I am using librealsense2. Should I rename the files? There is a librealsense without the '2' the '1' version?

Kuchiriel, 2018-07-18:
No success, same error, tryed the option from AUR, no success either.

toinsson, 2018-07-19:
This wrapper is not compatible with librealsense2. Intel did develop their own python extension - named pyrealsense2 - which you might be able to pull from pip.

closed 2018-07-19 by toinsson

swjtuccy, 2019-07-22:
> @toinsson Same location but I've seen I am using librealsense2. Should I rename the files? There is a librealsense without the '2' the '1' version? I will try with this https://github.com/IntelRealSense/librealsense/releases/tag/v1.9.7 and give a feedback.
did u make it to import pyrealsense2 on tx2?