toinsson/pyrealsense #83: Issue with LRS 1.12.1 and PRS 2, R200 & F200 devices not detected on UP2

AdamMiltonBarker, 2018-10-03:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 | 
| Firmware Version | 4.16.0-041600-generic | 
| Operating System & Version |  Ubuntu 16.04 |
| librealsense version | 1.12.1  |
| pyrealsense version |pyrealsense2  |

Hi guys I have a closed issue that is not resolved on Librealsense but I am thinking possibly the issue maybe on PyRealSense side, I am talking with a buddy at Intel also to try resolve this. 

Issue is on 2 separate UP2 devices, the legacy version of librealsense 1.12.1 installs fine, as does pyrealsense2, I no longer get the error from pyrealsense now that I am using what I presume to be the legacy version, but the  both F & R200 cams are detected by the UP2 itself (shows fine in dmesg) but is not detected when running the sample from PyPi, related issue and screen shots of installation below:

https://github.com/IntelRealSense/librealsense/issues/1640

[image]

[image]

toinsson, 2018-10-04:
Just a reminder, `pyrealsense2` has no connection with `pyrealsense` (as it is generated directly from the source of `librealsense` with pybind11) and `pyrealsense` only works with `librealsense` v1.x.

If your issue is not `pyrealsense` related, then I am afraid that here is not the place to get a solution.

AdamMiltonBarker, 2018-10-04:
Hi please see the attached issue and you will see the issue originated from
using pyrealsense, once I swapped to pyrealsense2 with legacy librealsense
1.12.1 there are no longer errors but the cameras are still not detected

-- 

Regards,

*Adam Milton-Barker*
*BigFinite IoT Engineer*
*Intel Software Innovator*

Website <https://adammiltonbarker.com/>, Google+
<https://plus.google.com/+AdamMiltonBarker>, LinkedIn
<https://es.linkedin.com/in/adammiltonbarker>, Twitter
<https://www.twitter.com/AdamMiltonBarkr>, Facebook
<https://www.facebook.com/AdamMiltonBarker>
TEL:  <https://www.facebook.com/AdamMiltonBarker>+34 603 358 245

toinsson, 2018-10-04:
Ok, indeed R200 and F200 are only supported by librealsense v1.x so you will need to use pyrealsense (not pyrealsense2).

In the log you provided there are a couple of warnings:

`
import pyrealsense
/usr/local/lib/python3.5/dist-packages/pyrealsense/extlib.py:26: UserWarning: librealsense not found.
  warnings.warn("librealsense not found.")
/usr/local/lib/python3.5/dist-packages/pyrealsense/extlib.py:33: UserWarning: rsutilwrapper not found.
  warnings.warn("rsutilwrapper not found.")
`
seems like "librealsense not found". In case you installed it elsewhere than in default location, you can use PYRS_INCLUDES and PYRS_LIBS to indicate the correct location.

AdamMiltonBarker, 2018-10-04:
Hi thanks ok yer pyrealsense has the issue but everything was installed in
default location I followed the guide exactly. Not sure where I read
pyrealsense2 should be used but will try your suggestion tonight this would
make sense and hopefully is the cause of the issue

-- 

Regards,

*Adam Milton-Barker*
*BigFinite IoT Engineer*
*Intel Software Innovator*

Website <https://adammiltonbarker.com/>, Google+
<https://plus.google.com/+AdamMiltonBarker>, LinkedIn
<https://es.linkedin.com/in/adammiltonbarker>, Twitter
<https://www.twitter.com/AdamMiltonBarkr>, Facebook
<https://www.facebook.com/AdamMiltonBarker>
TEL:  <https://www.facebook.com/AdamMiltonBarker>+34 603 358 245

AdamMiltonBarker, 2018-10-04:
OK so not sure what it was in the end, but all is working :) 1.12.1 and PyRealSense. Thanks for the help guys. I feel Github should so some stripping when parsing emails lol. 

[image]

AdamMiltonBarker, 2018-10-12:
Sorry @toinsson  I am going to reopen this one as it is not resolved, which I found when actually attempting to read. There are two of us discussing the fix for this, a friend is helping work it out who has experience with using UP2 and Realsense legacy cams, and we are stumped I believe.

1. I had an issue shown in the code block below, I found downgrading numpy may work, this is specifically f200, I did this and there was no longer the error below, but the camera light does not come on and the captured frames are entirely black.

```
 File "/home/upsquared/.local/lib/python3.5/site-packages/numpy/ctypeslib.py", line 183, in check_retval
    return array(self)
ValueError: '<P' is not a valid PEP 3118 buffer format string
```

2. R200, this gives the following errors, I have attempted downgrading to kernal 4.13 as I found suggested, when selecting this kernal the UP2 will not load (not your issue) and here are tests I made, the next highest suggested kernal is 4.10 which would put me back at the very beginning as upgrading from 4.10 to 4.16 resolved me initial issue:

```
modinfo uvcvideo | grep version
version:        1.1.1
srcversion:     F7FCC3FC6FFF80DA00B9B92

dmesg -wT

[Thu Oct 11 15:09:41 2018] usb 1-2: USB disconnect, device number 2
[Thu Oct 11 15:09:41 2018] usb 1-2.2: USB disconnect, device number 3
[Thu Oct 11 15:09:41 2018] usb 2-2: USB disconnect, device number 2
[Thu Oct 11 15:09:41 2018] usb 2-2.1: USB disconnect, device number 7
[Thu Oct 11 15:09:51 2018] usb 2-3: new SuperSpeed USB device number 8 using xhci_hcd
[Thu Oct 11 15:09:51 2018] usb 2-3: New USB device found, idVendor=8086, idProduct=0a80
[Thu Oct 11 15:09:51 2018] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Oct 11 15:09:51 2018] usb 2-3: Product: Intel RealSense 3D Camera R200
[Thu Oct 11 15:09:51 2018] usb 2-3: Manufacturer: Intel Corp
[Thu Oct 11 15:09:51 2018] usb 2-3: SerialNumber: SN_2221001899
[Thu Oct 11 15:09:51 2018] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Thu Oct 11 15:09:51 2018] uvcvideo 2-3:1.0: Entity type for entity Extension 2 was not initialized!
[Thu Oct 11 15:09:51 2018] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[Thu Oct 11 15:09:51 2018] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:15.0/usb2/2-3/2-3:1.0/input/input11
[Thu Oct 11 15:09:51 2018] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Thu Oct 11 15:09:51 2018] uvcvideo: Unable to create debugfs 2-8 directory.
[Thu Oct 11 15:09:51 2018] uvcvideo 2-3:1.2: Entity type for entity Camera 4 was not initialized!
[Thu Oct 11 15:09:51 2018] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:15.0/usb2/2-3/2-3:1.2/input/input12
[Thu Oct 11 15:09:51 2018] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Thu Oct 11 15:09:51 2018] uvcvideo: Unable to create debugfs 2-8 directory.
[Thu Oct 11 15:09:51 2018] uvcvideo 2-3:1.4: Entity type for entity Processing 7 was not initialized!
[Thu Oct 11 15:09:51 2018] uvcvideo 2-3:1.4: Entity type for entity Camera 6 was not initialized!
[Thu Oct 11 15:09:51 2018] input: Intel RealSense 3D Camera R200: as /devices/pci0000:00/0000:00:15.0/usb2/2-3/2-3:1.4/input/input13
[Thu Oct 11 15:29:23 2018] usb 1-4: USB disconnect, device number 4
[Thu Oct 11 15:29:24 2018] usb 2-4: new SuperSpeed USB device number 9 using xhci_hcd
[Thu Oct 11 15:29:24 2018] usb 2-4: New USB device found, idVendor=03e7, idProduct=f63b
[Thu Oct 11 15:29:24 2018] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Oct 11 15:29:24 2018] usb 2-4: Product: VSC Loopback Device
[Thu Oct 11 15:29:24 2018] usb 2-4: Manufacturer: Movidius
[Thu Oct 11 15:29:24 2018] usb 2-4: SerialNumber: 5CB83DCF94E20B

lsusb
Bus 002 Device 009: ID 03e7:f63b
Bus 002 Device 008: ID 8086:0a80 Intel Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Again sorry to revive but I really am pushing to find a solution, next week I demo the github project at my workplace and I would like and actually need the realsense to be working lol, advice would be really cool guys, thanks in advance.

AdamMiltonBarker, 2018-10-13:
@toinsson I had found that which is what prompted me to the fix of Numpy 1.14.4 downgrade, the issue with the first is additional to that issue, ie I applied the fixes I found related that issue and it has solved the error but now have additional issues, the frames are totally black, the device connects fine (f200), no errors on reading the frame, but light does not come on on camera (which is  suppose to happen as soon as the application opens it right ? and the frame is black, so I believe the issue you linked to be solved in my case, but I have additional issues where the connection is saying successfully where it might not actually be? 

RE 2 would be cool if you could explain why you feel that, it may help me narrow down possible solutions, thanks for the reply and help!

pzoltowski, 2019-03-08:
@AdamMiltonBarker I'm struggling with my R200 on mac as well. Did you manage to sort it out?

closed 2020-08-17 by toinsson