toinsson/pyrealsense #90: Installation Problem

MiguelKoch, 2019-12-18:
| Camera Model | F200 | 
| Firmware Version | - | 
| Operating System & Version | Windows 10 1909  |
| librealsense version | 1.12.1 |
| pyrealsense version |  2.2 |

Hi, I'm having path problems (path nightmares, more like it)

I have librealsense correctly installed. I can run all the examples at least... However, i dont have the /usr/local/include/librealsense directory. Instead, my headers are located in -build\include\librealsense.
I was getting the "Cannot open include file: 'rs.h': No such file or directory" error. I backed up setup.py and messed around, changing the directories to the correct ones and i now it can find rs.h but it still can't find rsutil.h.

________________________________________________________________________________________________________
________________________________________________________________________________________________________

running install
running bdist_egg
running egg_info
writing pyrealsense.egg-info\PKG-INFO
writing dependency_links to pyrealsense.egg-info\dependency_links.txt
writing requirements to pyrealsense.egg-info\requires.txt
writing top-level names to pyrealsense.egg-info\top_level.txt
reading manifest file 'pyrealsense.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pyrealsense.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'pyrealsense.rsutilwrapper' extension
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ipyrealsense -IC:\Anaconda3\envs\OpenPose\lib\site-packages\numpy\core\include "-IC:\Users\ISR 001\Documents\GitHub\librealsense-1.12.1\build-vc15\include\librealsense" "-IC:\Users\ISR 001\Documents\GitHub\pyrealsense\docs" -IC:\Anaconda3\envs\OpenPose\include -IC:\Anaconda3\envs\OpenPose\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyrealsense/rsutilwrapper.cpp /Fobuild\temp.win-amd64-3.7\Release\pyrealsense/rsutilwrapper.obj
rsutilwrapper.cpp
C:\Anaconda3\envs\OpenPose\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ipyrealsense -IC:\Anaconda3\envs\OpenPose\lib\site-packages\numpy\core\include "-IC:\Users\ISR 001\Documents\GitHub\librealsense-1.12.1\build-vc15\include\librealsense" "-IC:\Users\ISR 001\Documents\GitHub\pyrealsense\docs" -IC:\Anaconda3\envs\OpenPose\include -IC:\Anaconda3\envs\OpenPose\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tppyrealsense/rsutilwrapperc.cpp /Fobuild\temp.win-amd64-3.7\Release\pyrealsense/rsutilwrapperc.obj
rsutilwrapperc.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:/usr/local/lib "/LIBPATH:C:\Users\ISR 001\Documents\GitHub\librealsense-1.12.1\build-vc15" /LIBPATH:C:\Anaconda3\envs\OpenPose\libs /LIBPATH:C:\Anaconda3\envs\OpenPose\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\um\x64" realsense.lib /EXPORT:PyInit_rsutilwrapper build\temp.win-amd64-3.7\Release\pyrealsense/rsutilwrapper.obj build\temp.win-amd64-3.7\Release\pyrealsense/rsutilwrapperc.obj /OUT:build\lib.win-amd64-3.7\pyrealsense\rsutilwrapper.cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\pyrealsense\rsutilwrapper.cp37-win_amd64.lib
LINK : fatal error LNK1181: cannot open input file 'realsense.lib'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.24.28314\\bin\\HostX86\\x64\\link.exe' failed with exit status 1181

toinsson, 2019-12-18:
from 
```
 LINK : fatal error LNK1181: cannot open input file 'realsense.lib'
```
seems like you forgot to set PYRS_LIBS.

MiguelKoch, 2019-12-19:
Thank you for your quick reply and your time.

Right now, **inside setup.py** my **inc_dirs** list contains "np.get_include()", the path to the librealsense headers ("-\build-vc15\include\librealsense") and **PYRS_INCLUDES** has the path to the folder containing the pyrealsense rs.h file (-\pyrealsense\docs).

The **lib_dirs** list contains "/usr/local/lib" (which as far as i know does nothing ie. I don't have a '/usr/local/lib') and for **PYRS_LIBS** I have "-\librealsense-1.12.1\build-vc15", which is where CMAKE built librealsense.

A few of these are probably wrong I assume. Should i set **PYRS_LIBS** to '\build-vc15\lib' which is where realsense.lib is located? I have a backup of the original setup.py as well.

_________________________________________________________________________________________________________
_________________________________________________________________________________________________________

**What's inside each directory** (I don't know if this will make it easier for you or more confusing)

**-\include\librealsense:**
             ---- rs.h
             ---- rs.cpp
             ---- rscore.hpp
             ---- rsutil.h

**-\pyrealsense\docs:**
             ---- rs.h
             ---- pyrealsense.rst
             ---- index.rst
             ---- ...
             ---- ...

**-\librealsense-1.12.1\build-vc15:**
             ----  bin folder
             ----  CMakeFiles folder
             ----  include folder
             ----  A few more folders ( lib, realsense.dir, release, unit-tests, x64)
             ----  Realsense.snl and a bunch of .vcxproj files

**-'\build-vc15\lib':**
             ----  realsense.lib

closed 2020-08-17 by toinsson

GiantDeveloper021, 2021-09-16:
hello i hope you are doing well i am stuck in same issue did you solve the issue or not ? if so ,can you please guide me to how to solve this problem. Thanks

MiguelKoch, 2021-09-17:
Hi!

I wish I could help you, but unfortunately I was never able to solve this
issue. Shortly after I raised the ticket my company brought in new cameras
so I abandoned the issue. Best of luck!